arduino/Arduino #5317: https://www.arduino.cc/en/Tutorial/LibraryExamples doesn't show all tutorial pages

q2dg, 2016-08-30:
Specifically, I was looking for the Curie's tutorials which are only listed at bottom of https://www.arduino.cc/en/Reference/CurieBLE . I think these tutorials should be place (also) in LibraryExamples page as a central entry point, as long as CurieIMU, CurieTimer etc examples.

agdl, 2016-08-31:
@00alis

q2dg, 2016-09-02:
Tutorials listed in https://www.arduino.cc/en/Guide/ArduinoZero#toc6 aren't in LibraryExamples neither.

00alis, 2016-09-05:
Adding @SimonePDA and @akash73 who are taking care of the documentation pages.
@q2dg thank you so much for reporting the issue!

q2dg, 2017-07-09:
Well, still waiting.
Meanwhile,  https://www.arduino.cc/en/Guide/ArduinoZero#toc6 has become  https://www.arduino.cc/en/Guide/ArduinoZero#toc7

akash73, 2017-07-17:
hello @q2dg ,
thanks for bring up again.
we decided to keep the two types of example in different areas: the BLE IMU and others come with the core and is not a separate library.
in https://www.arduino.cc/en/Tutorial/LibraryExamples we list all the examples that you'll find by downloading the library.
Other examples fall in proper reference page like : 

https://www.arduino.cc/en/Reference/CurieIMU
https://www.arduino.cc/en/Reference/CurieBLE 
https://www.arduino.cc/en/Reference/CurieTimerOne

per1234, 2017-07-17:
It sounds like this issue is a wontfix and @q2dg gave a thumbs up so I'm going to close it. Feel free to re-open (or request a re-open) if I'm wrong about this.

closed 2017-07-17 by per1234