arduino/Arduino #5318: [Library manager]KTM-S1201 LCD library

MCUdude, 2016-08-30:
Please add this library to the library manager. It's an LCD library for all the KTM-S1201 LCDs out there!
[https://github.com/MCUdude/KTMS1201](https://github.com/MCUdude/KTMS1201)

agdl, 2016-08-31:
I'm sorry, but your library can't be added since a tag is missing please follow [this guide](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ)

MCUdude, 2016-08-31:
I'll fix this ASAP!

MCUdude, 2016-08-31:
I've done some changes to the library.properties file, and drafted a release. Does the library manager fetch the latest version of the library as soon as I draft a new release, or du we have to go through all this again?

agdl, 2016-08-31:
It is an automatic process. Once there is the github repo entry all new releases will be automatically added.

However this is wrong

https://github.com/MCUdude/KTMS1201/blob/v1.0.1/library.properties

tag is 1.0.1 and version=1.0.0

MCUdude, 2016-08-31:
damn it! Is there any way of deleting the 1.0.1 release and "re release it" with the fix in the properties file, or do I need to bump the version up to v1.0.2 or v1.0.1r1?

agdl, 2016-08-31:
of course but I know the command line way:

```
cd path to the repo
git tag -d 1.0.1 
git push origin :1.0.1
```

Now you deleted the tag then you have to push it again (fix it before :) )

MCUdude, 2016-08-31:
I drafted a release and didn't create a tag. When I'm running `git tag -d 1.0.1` i get this error:

´error: tag '1.0.1' not found.´

Is there any way I can delete or edit the release itself?

agdl, 2016-08-31:
sorry there is a v.. so 
```
 	
v1.0.1 
```

MCUdude, 2016-08-31:
Of course! I think everything should be OK now 😃

agdl, 2016-09-01:
Done thank you!

closed 2016-09-01 by agdl