arduino/Arduino #5322: Code Folding on switch-case

MarcelloTheArcane, 2016-08-31:
With code folding enabled, I think it would be good to be able to fold the cases within a 'switch'

matthijskooijman, 2016-08-31:
I believe the folding is taken care of by the RSyntaxTextArea that we use to display the code, so that would be something to add there. Perhaps you could file a issue with them? I suspect that this is not something that would have a priority for the Arduino developers, and even if it did they would probably want to submit a patch upstream and use that rather than adding this just for Arduino.

MarcelloTheArcane, 2016-08-31:
umm.. sorry to be annoying, but how do i do that? where is RSyntaxTextArea?

matthijskooijman, 2016-08-31:
> umm.. sorry to be annoying, but how do i do that? where is RSyntaxTextArea?
Did you try googling for it? There is a git repository for it, but I'd
have to google to get you the link...

Gr.

Matthijs

cmaglie, 2016-08-31:
https://github.com/bobbylight/RSyntaxTextArea

per1234, 2017-07-06:
@MarcelloTheArcane's feature request on the RSyntaxTextArea repository: https://github.com/bobbylight/RSyntaxTextArea/issues/199

It is already possible to accomplish this by wrapping the contents of each case in braces:
```arduino
switch (foo) {
case 1:
  {
    // code for handling case 1
    break;
  }
case 2:
  {
    // code for handling case 2
    break;
  }
default:
  {
    // code for handling default case
    break;
  }
}
```
[image]
This also solves "crosses initialization of" errors. Of course you can leave the `break` statements outside the braces if you prefer to be able to see if any of the cases fall through without unfolding them.