arduino/Arduino #5323: Arduino IDE 1.6.10 (and 1.6.11) crash on OS X 10.11.6

alexodus, 2016-08-31:
This is the exception:
```
$./Arduino.app/Contents/MacOS/Arduino 
java.lang.NullPointerException
	at processing.app.Base.lambda$static$2(Base.java:85)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:174)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
	at java.util.stream.ReferencePipeline$2$1.accept(ReferencePipeline.java:175)
	at java.util.LinkedList$LLSpliterator.forEachRemaining(LinkedList.java:1235)
	at java.util.stream.AbstractPipeline.copyInto(AbstractPipeline.java:481)
	at java.util.stream.AbstractPipeline.wrapAndCopyInto(AbstractPipeline.java:471)
	at java.util.stream.ReduceOps$ReduceOp.evaluateSequential(ReduceOps.java:708)
	at java.util.stream.AbstractPipeline.evaluate(AbstractPipeline.java:234)
	at java.util.stream.ReferencePipeline.collect(ReferencePipeline.java:499)
	at processing.app.Base.getIDELibs(Base.java:1062)
	at processing.app.Base.rebuildExamplesMenu(Base.java:1163)
	at processing.app.Editor.buildFileMenu(Editor.java:575)
	at processing.app.Editor.buildMenuBar(Editor.java:465)
	at processing.app.Editor.<init>(Editor.java:252)
	at processing.app.Base.handleOpen(Base.java:829)
	at processing.app.Base.restoreSketches(Base.java:519)
	at processing.app.Base.<init>(Base.java:457)
	at processing.app.Base.guardedMain(Base.java:223)
	at processing.app.Base.main(Base.java:136)
```

```
$ java -version
java version "1.8.0_92"
Java(TM) SE Runtime Environment (build 1.8.0_92-b14)
Java HotSpot(TM) 64-Bit Server VM (build 25.92-b14, mixed mode)
```

with  1.6.9 there aren't problems.

facchinm, 2016-08-31:
Hi @alexodus , that's quite peculiar. I just tested the nightly on 10.11.6 and it works correctly. Does the IDE crash also if you move the app to another location?

PaulStoffregen, 2016-08-31:
FWIW, if #5178 gets re-merged, it'll replace this complicated lamba filter stuff with a simple loop that iterates over the libraries to rebuild the examples menu.

Any chance you could give test build "PR-5178-BUILD-585" a try on your Mac?
https://github.com/arduino/Arduino/pull/5178#issuecomment-236925862

alexodus, 2016-08-31:
I'm used these versions today on downloads directory ed in /Applications directory. Versions tested are: 1.6.10, 1.6.11, 1.6.11 nightly

vicnevicne, 2016-09-03:
FWIW, I also had a systematic crash on IDE startup on Windows, due to a board X being selected in the IDE on exit, then the cores with that board X being deleted from disk. 
This crash is blocking as the IDE cannot be restarted to select another board. The preference file has to be modified manually but valid values of target_package and target_platform are not obvious.
The root cause is the COMPATIBLE Predicate which is not protected against BaseNoGui.getTargetPlatform() returning null in those conditions.
@PaulStoffregen Looks like your rewrite of this logic in #5178 handles it OK, however the Arduinobot build referenced above is now 404 , and this PR is not merged in the trunk yet if I'm not mistaken.
Is there a build with this change available sowewhere so I can confirm it is OK ?

jmurphy644, 2016-09-07:
I made the mistake of reinstalling the IDE to fix a different problem and I now have this same issue under windows 10.  (1.6.11)

xoration, 2016-09-18:
Is there a solution for this issue yet? Or is the only options downgrading?

cmaglie, 2016-09-22:
This should be solved in IDE 1.6.12 with #5386.
Please reopen if the issue persists.

closed 2016-09-22 by cmaglie