arduino/Arduino #5324: [Library Manager] Please add the Arduino NanoPlayBoard library

josejuansanchez, 2016-08-31:
https://github.com/josejuansanchez/NanoPlayBoard-Arduino-Library

Thanks! :)

per1234, 2016-09-01:
The library needs to be in the root of your repository, not in a subfolder as it is now.

From https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#is-my-git-repo-ok:
> Your repo is ok if its root folder contains file library.properties and the rest of the library source code. If your library is stored into a subfolder, your repo it's not ok and we can't include your library in the list.

josejuansanchez, 2016-09-01:
I have fixed the issue. I have moved the library source code to the root folder.

Could you please review the library?

Thanks!

agdl, 2016-09-02:
@josejuansanchez why you have 2 folders for examples? Now you have:
* sketches
* examples
you should move all the .ino files in the ```example``` folder

josejuansanchez, 2016-09-02:
@agdl I would like to maintain the two folders if it could be possible.

* The `examples` folder includes basic examples about how to use the library.
* The `sketches` folder includes experimental projects **«a bit more complex»** and not finished yet These experimental projects can be moved to the examples folder when they are finished.

agdl, 2016-09-02:
You can have them, but only the one contained in ```example``` will be showed in the IDE menù

agdl, 2016-09-02:
maybe you can rename the ```sketches``` folder in ```experimental projects``` and place it in the ```example``` folder so you have both

josejuansanchez, 2016-09-02:
@agdl That's the point, I want to show in the IDE menu only the `examples` folder.

I would like continue developing the experimental projects in the current folder and show them in the Arduino IDE only when they are finished.

Could it be possible?

agdl, 2016-09-02:
Of course. However you have to tag the repo at current status to be added

josejuansanchez, 2016-09-02:
The current status of the repo is tagged as **v0.0.2-alpha**. Is it enough?

https://github.com/josejuansanchez/NanoPlayBoard-Arduino-Library/tags

agdl, 2016-09-02:
https://github.com/josejuansanchez/NanoPlayBoard-Arduino-Library/tree/v0.0.2-alpha/library.properties

No the current tag refer to a status without a library.properties in this position

josejuansanchez, 2016-09-02:
The repository has been tagged as **v0.0.2** and now seems that everything work fine :)

https://github.com/josejuansanchez/NanoPlayBoard-Arduino-Library/blob/v0.0.2/library.properties

agdl, 2016-09-05:
Done thank you!

closed 2016-09-05 by agdl

josejuansanchez, 2016-09-05:
@agdl Thanks! :)