arduino/Arduino #5325: ESP8266

iliemv, 2016-08-31:
Arduino: 1.6.11 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Board nodemcuv2 (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

cmaglie, 2016-08-31:
You're using a 3rd party core that is not provided by Arduino. Please post your question (?) on the correct issue tracker.

closed 2016-08-31 by cmaglie

facchinm, 2016-09-01:
@iliemv , you can solve the problem by manually removing the old core which did not uninstall correctly (follow instruction here https://github.com/arduino/Arduino/issues/5231#issuecomment-239385304 changing `Sparkfun` into `esp8266` )
See https://github.com/arduino/arduino-builder/pull/84