arduino/Arduino #5330: HWSPI initializes and sets D10 high, even if not used

tekka007, 2016-08-31:
HWSPI assumes CSN/SS pin on **D10** and initializes it.

- static SS pin assignment to **D10**: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L38 
and https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L43

- Pin **D10** is defined as output and set high upon SPI initialization: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/SPI/src/SPI.cpp#L31-L45

This is an unwanted behavior if alternative CSN/SS and CE pins are used.

matthijskooijman, 2016-09-01:
This is a limitation of the AVR SPI hardware - if the SS pin is an INPUT and driven low (or something like that), the SPI hardware will switch into slave mode. Making it an output makes sure this does not happen. The fact that you're surprised about this does suggest this should be better documented perhaps?

tekka007, 2016-09-03:
Yes, you're right!
The documentation (in the SPI lib) should probably be updated accordingly.

closed 2016-09-03 by tekka007

q2dg, 2016-09-06:
@agdl maybe?