arduino/Arduino #5332: Variable expansion in platfrom.txt seems inconsistent

beikeland, 2016-09-02:
Background: trying to make 3rd part hw spec for using esp8266avrisp as programmer.

Define new programmer.txt with `8266avrisp.program.extra_params=-Pnet:{serial.port}:328` and `8266avrisp.program.extra_filename=.with_bootloader`

Define new tool:`tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}{program.extra_filename}.hex:i"`

Output: `C:/temp/avrdude-net/avrdude -CC:\arduino-1.6.11\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -carduino -P10.10.0.185 {program.extra_params} -b115200 -D -Uflash:w:C:\...../avr_echo.ino{program.extra_filename}.hex:i`

`{program.extra_params}` and `{program.extra_filename}` are not expanded when using "Upload using programmer" They are expanded correctly when burning the bootloader. Just added `{program.extra_params}` at the end to see of anything i defined was usable in the tool definition. It was.

`tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params}  -e -Ulock:w:{bootloader.unlock_bits}:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m {program.extra_filename}`

Output: `c:/temp/avrdude-net/avrdude -CC:\arduino-1.6.11\hardware\tools\avr/etc/avrdude.conf -v -patmega328p -cstk500v1 -Pnet:10.10.0.185:328 -cstk500v1 -e -Ulock:w:0x3F:m -Uefuse:w:0x05:m -Uhfuse:w:0xDE:m -Ulfuse:w:0xFF:m .with_bootloader`

It also seems like its neccecary to create a new board.txt entry with a duplicate entry and `program.tool` set to a new tool, instead of overriding the tool in programmers.txt. This could be by design, but it would mean i would have to create two tool entries in platform.txt instead of adding a paramteter `8266avrisp.program.extra_filename=.with_bootloader` to programmers.txt to allow downloading of sketch with or without the bootloader.

I was hoping I would not need to create new boards or tools (except temporary a new tool as avrdude.exe doesn't work well with sockets on windows at the moment). But it seems I need to create a new board to use the new tool, and that the parameters from programmer is not expanded properly except when burning the bootloader.

Would it be possible to override the tool path in programmer.txt without changing platform.txt, removing the need to duplicate a lot of the standard avr core?

boards.txt
```
menu.cpu=Processor

uno.name=Arduino/Genuino Uno-esp8266
uno.upload.via_ssh=true

uno.upload.tool=avrdude
uno.upload.programmer=8266avrisp
uno.program.programmer=8266avrisp
uno.upload.protocol=arduino
uno.upload.maximum_size=32256
uno.upload.maximum_data_size=2048
uno.upload.speed=115200

uno.bootloader.tool=avrdude
uno.bootloader.low_fuses=0xFF
uno.bootloader.high_fuses=0xDE
uno.bootloader.extended_fuses=0x05
uno.bootloader.unlock_bits=0x3F
uno.bootloader.lock_bits=0x0F
uno.bootloader.file=optiboot/optiboot_atmega328.hex

uno.build.mcu=atmega328p
uno.build.f_cpu=16000000L
uno.build.board=AVR_UNO
uno.build.core=arduino
uno.build.variant=standard
````
programmers.txt
```
8266avrisp.name=8266avrisp
8266avrisp.communication=serial
8266avrisp.protocol=stk500v1
8266avrisp.program.protocol=stk500v1
8266avrisp.program.tool=avrdude
8266avrisp.program.extra_params=-Pnet:{serial.port}:328 -c{program.protocol}
8266avrisp.program.extra_filename=.with_bootloader
```
platform.txt
```
name=ESP8266AVRISP
version=0.0.1
# AVR compile variables
# ---------------------

compiler.warning_flags=-w
compiler.warning_flags.none=-w
compiler.warning_flags.default=
compiler.warning_flags.more=-Wall
[... 70 lines cut ...]
preproc.macros.flags=-w -x c++ -E -CC
recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} {preproc.macros.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{preprocessed_file_path}"

tools.avrdude.path=c:/temp/avrdude-net
tools.avrdude.cmd.path={path}/avrdude
tools.avrdude.config.path={runtime.tools.avrdude.path}/etc/avrdude.conf

tools.avrdude.upload.params.verbose=-v
tools.avrdude.upload.params.quiet=-q -q
tools.avrdude.upload.params.noverify=-V
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} {program.extra_params} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i" {upload.extra_filename}
            
tools.avrdude.program.params.verbose=-v
tools.avrdude.program.params.quiet=-q -q
tools.avrdude.program.params.noverify=-V
tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}{program.extra_filename}.hex:i" 

tools.avrdude.erase.params.verbose=-v
tools.avrdude.erase.params.quiet=-q -q
tools.avrdude.erase.pattern="{cmd.path}" "-C{config.path}" {erase.verbose} -p{build.mcu} -c{protocol} {program.extra_params}  -e -Ulock:w:{bootloader.unlock_bits}:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:{bootloader.high_fuses}:m -Ulfuse:w:{bootloader.low_fuses}:m {program.extra_filename}

tools.avrdude.bootloader.params.verbose=-v
tools.avrdude.bootloader.params.quiet=-q -q
tools.avrdude.bootloader.pattern="{cmd.path}" "-C{config.path}" {bootloader.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{runtime.platform.path}/bootloaders/{bootloader.file}:i" -Ulock:w:{bootloader.lock_bits}:m
```

facchinm, 2016-09-06:
Hi @beikeland,
I believe you are making things too complicated :smile: 
The programmers defined in programmers.txt can only be used when selecting "Upload using programmer". In your use case, the "programmer" is more a collection of avrdude configs, which can be passed by the board. I suggest doing something like

In boards.txt inherit everything possible from arduino core
```
menu.cpu=Processor

uno.name=Arduino/Genuino Uno-esp8266
uno.upload.via_ssh=true

uno.upload.tool=avrdudenet
uno.upload.programmer=8266avrisp
uno.program.programmer=8266avrisp
uno.upload.protocol=arduino
uno.upload.maximum_size=32256
uno.upload.maximum_data_size=2048
uno.upload.speed=115200

uno.bootloader.tool=avrdudenet
uno.bootloader.low_fuses=0xFF
uno.bootloader.high_fuses=0xDE
uno.bootloader.extended_fuses=0x05
uno.bootloader.unlock_bits=0x3F
uno.bootloader.lock_bits=0x0F
uno.bootloader.file=optiboot/optiboot_atmega328.hex

uno.build.mcu=atmega328p
uno.build.f_cpu=16000000L
uno.build.board=AVR_UNO
uno.build.core=arduino:arduino
uno.build.variant=arduino:standard
```
(note that upload.tool is now `avrdudenet`)

In platform.txt, instead, we only need

```
name=ESP8266AVRISP
version=0.0.1
# AVR compile variables
# ---------------------

tools.avrdudenet.path={runtime.tools.avrdude.path}
tools.avrdudenet.cmd.path={path}/bin/avrdude
tools.avrdudenet.extra_params=-Pnet:{serial.port}:328 -cstk500v1
tools.avrdudenet.extra_filename=with_bootloader
tools.avrdudenet.config.path={runtime.tools.avrdude.path}/etc/avrdude.conf

tools.avrdudenet.upload.params.verbose=-v
tools.avrdudenet.upload.params.quiet=-q -q
tools.avrdudenet.upload.params.noverify=-V
tools.avrdudenet.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -P{serial.port} {extra_params} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.{extra_filename}hex:i"
```

This should do the trick

beikeland, 2016-09-06:
Interesting, I've been banging my head against the wall for a while with this. 

But my goal isn't actually to create a new board, it is to define a new programmer from where I'm sitting. Well two new progarmmers, to be used with upload using programmer. One through the bootloader with esp-link firmware, and one ISP with the esp8266avrisp firmware.

I'd like that to be available to all arduino type boards without redefining all of them. Allowing the programmers.txt tool entry to override the tool entry from boards.txt would go a long way towards achieving that?

Unknown why at some point I started getting network upload using programmer is not supproted, and the build process correctly expanded the variables. Adding mDNS txt upload_ssh=no to the programmers firmware solved that, but I would rather define a tool and a programmer, than a tool and many boards if that makes sense?

facchinm, 2016-09-06:
I understand what you want to achieve, but right now every platform can only use its own programmers, not externally defined one. 

You could shadow avr core while in development, eventually opening a PR to include your "programmer" in mainline when it's production ready.

In the meantime, if you think that other people could find it useful, feel free to propose an architecture change in the [dev mailing list](https://groups.google.com/a/arduino.cc/forum/#!forum/developers)

beikeland, 2016-09-06:
To include the programmers mainline down the road or changing it so programmers can override a tool definition both solves what I want to achieve. And I think the use of esp8266 moduels to to wireless uploads will be usefull.

But thinking ahead, if an 3rd party hardware designer makes new programmer, what would be the best way for him to publish a json file to allow his programmer to be used with existing boards in the future? Creating a board is straight forward, a programmer for existing boards not so much.

I think I'll settle for keep at working out the kinks and try to have both esp-link and 8266avirsp added to the main code, and hopefully avrdude will be able to to network uploads on windows by then. I don't have the time to drive an architecture change, but wanted to see which way would be the best way.

Thanks for your input! I guess this can be closed as the problem it initially describes went away all on its own.

closed 2016-09-06 by beikeland

per1234, 2016-09-25:
It is possible to use the Arduino IDE's 3rd party hardware system to install programmers, though there are some limitations. I have two of them I made that can be installed via Boards Manager([ATmega32U4asISP](https://github.com/per1234/ATmega32U4asISP) and [Arduino-AVRISPmkII-fix](https://github.com/per1234/Arduino-AVRISPmkII-fix)). I also attempted to make one for the ArduinoISP.org programmer(yeah I know, .org boo hiss, but I feel bad for the users who unknowingly buy one and can't use it with the .cc IDE, even though I'm not one of them) but was stymied by not being able to specify the path to the required custom avrdude.conf.

Although a less common usage, I think supporting the installation of 3rd party programmers could be a powerful feature of the Arduino IDE. I'd like to see the support for this usage improved, though I'm not sure how this could be done without unintended side effects.

beikeland, 2016-09-25:
What I'm seeing is there is a difference how "Burn bootloader" and "Upload
using programmer" is handled.

To me it seems a 3rd party programmer will work as I expected it to with
"Burn bootloader", but not "Upload using programmer" and I also seem to
experience inconsistencies I cannot explain, so I probably need to spend
more time figuring out whats what to make sense of it. But it seems to me
if the uploads was handled the same as the bootloader it would be all
hakuna matata.

per1234, 2016-09-25:
I've also encountered that issue. The platform.txt associated with the selected board is used for **Burn Bootloader** but the platform.txt associated with the selected programmer is used for **Upload Using Programmer**. The `{config.path}` value in platform.txt overrides the value set in programmers.txt during **Upload Using Programmer** but not during **Burn Bootloader**.  I'm not sure if this behavior was intended or not. If it is it would be nice to have better documentation.

It seems like changing things to be consistent will likely break something either way it's done. For example: Due to having a modified platform.txt, my Arduino-AVRISPmkII-fix programmer benefits from the current **Upload Using Programmer** behavior and the current **Burn Bootloader** behavior causes issues. However, some 3rd party boards(for example, some of the ones in [MightyCore](https://github.com/MCUdude/MightyCore)) require a custom avrdude.conf file and thus must use their own platform.txt. They benefit from the current **Burn Bootloader** behavior but the current **Upload Using Programmer** behavior causes them problems. This means the simple, obvious change of making the platform.txt selection behavior consistent will either benefit them but break my programmer or vice versa. If there's going to be a breaking change, sooner is better than later since the number of potentially affected projects will likely increase over time. The ideal would be to find a solution that works for all applications but this may not be possible. My programmer is probably used by very few, so breaking it won't be much of a concern but its requirements would likely be shared by other 3rd party programmers in the future.