arduino/Arduino #5339: Optiboot

PRosenb, 2016-09-03:
I converted the optiboot bootloader [optiboot_atmega328_pro_8MHz](https://github.com/Optiboot/optiboot/blob/master/optiboot/bootloaders/optiboot/optiboot_atmega328_pro_8MHz.hex) with convertHexToByteArray.py, replaced the file bootloader_lilypad328.h in Atmega_Board_Programmer with it and adjusted the start address in Atmega_Board_programmer on line 418 to 0x7e00.
Burning and verifying works afterwards but I am not able to upload a sketch to it using the 8Mhz breadboard board in the Arduino IDE. The LED on output 13 flashes three times quickly.
Any hints are appreciated.

NicoHood, 2016-09-03:
You need to file a bug at nick gammons github, not here. Why did you modify th end address?

PRosenb, 2016-09-03:
Absolutely, I mixed up the repos and will close this issue.

Still answering the question: It didn't work and it says that address in the converted h file.

closed 2016-09-03 by PRosenb