arduino/Arduino #5340: Adding Pololu 3pi Robot Library

sdale28, 2016-09-03:
Hi there. I just got a Pololu 3pi robot, and I found that the library they provide isn't compatible with the Arduino IDE. So I reorganized it, changed some code, and added a couple missing files  here: https://github.com/sdale28/Pololu-3pi-Library

I'm new to GitHub/Arduino but I think this would be a helpful library for people who buy the 3pi. Since I'm new, I'm not entirely sure how to make the library up to Arduino IDE standards so any advice would be greatly appreciated. 

Thanks!

per1234, 2016-09-04:
Since this is also a hardware package I recommend using Boards Manager to install it and bundle the library with the hardware package. That will be the easiest installation method for the users because Library Manager can't be used to install hardware.

See https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification for details on how to implement this.

sdale28, 2016-09-04:
Okay. Well, honestly, I'm not sure how to do a Boards Manager. Would the code still be useful as just a library without the hardware part?

per1234, 2016-09-04:
> Would the code still be useful as just a library without the hardware part?

From looking at your boards.txt it appears that it is a non-standard hardware configuration(20MHz and in one case ATmega48) therefore the hardware definitions must be installed since none of the Arduino AVR Boards can be used instead. In this case using Library Manager would actually increase the complexity of installation.

I just installed https://www.pololu.com/file/download/libpololu-arduino-150324.zip?file_id=0J868 from https://www.pololu.com/docs/0J17/3 and it seems to work fine with the latest IDE version and even bundles the libraries with the hardware files as I recommended. I don't understand what benefit your version provides over that one. If you can convince me I'd be willing to add Boards Manager support to your repository but this issue report is not the appropriate place to discuss it. If you want to continue please open an issue for the topic on your own repository and notify me of it,

closed 2016-09-04 by sdale28