arduino/Arduino #5341: Toolbar does not go full width on 4K

moxy-bc, 2016-09-04:
The Arduino toolbar does not display full width when maximized on a 4k screen.    See this screen-shot.

[image]

facchinm, 2017-06-26:
_@TheBetaFruit commented:_

> When using a 21:9 Ultra-Wide Monitor (e.g. the popular LG 34UM95, resolution = 3440x1440px), the upper turquise menu bar of the Arduino IDE is too short in full-screen mode and therefore centered with two ugly grey areas besides it.
> 
> It would be nice to fix this by making sure the menu bar stays left-aligned while expanding the turquise area to the right, fitting any screen width.
>

cmaglie, 2017-07-18:
Fix will be available on the next release 1.8.4

commit ddf60361 (2017-07-18): Toolbar can now go full width on 4K display

Fix #5341
closed 2017-07-18 by cmaglie