arduino/Arduino #5342: Arduino IDE 1.6.11 - SD card no access from within void loop()

billmaterial, 2016-09-04:
Moving from IDE 1.6.9 (working) to 1.6.11, I can access the SD card from void setup() but when entering void loop() the SD card cannot be accessed, this happens on the Windows and Linux (ubuntu) platforms.

Hardware: Arduino UNO + Maplin Touch LCD Shield for Arduino using TFT_Touch_Shield_V2-master library.
[tftbmp 1.6.11.ino.zip](https://github.com/arduino/Arduino/files/453927/tftbmp.1.6.11.ino.zip)

dlabun, 2016-09-04:
I think your problem is the TFT Touch library, it accesses the SPI bus and then never releases control... Adafruit had the same bug in their TFT library.

billmaterial, 2016-09-05:
Thank you for your comment dlabun, the TFT Touch library does work correctly on 1.6.9 but not on 1.6.11 which is strange, I will look to see if there is an update for it. The SD card can be accessed correctly from the setup routine multiple times with no problems, but it cannot be accessed when in the loop routine, it's seems once in the loop routine it locks out any access to the SD card.

per1234, 2021-03-28:
Hi @billmaterial Thanks for your interest in the SD library. At the current state of this report, using outdated software and with a far from minimal demonstration, I think this is a better fit for the Arduino Forum than this bug tracker.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code.

I'll keep an eye on that forum thread and if it is eventually determined to be caused by a bug in the current official software, I'll be sure to encourage a bug report to be opened, hopefully in the more appropriate repository this time around.

closed 2021-03-28 by per1234