arduino/Arduino #5345: Possible to add a "Refresh" or "Reload from Disk" option in toolbar?

gitmeister, 2016-09-05:
When editing outside of the IDE it would prevent having to close/re-open the Sketch since the IDE doesn't prompt to reload a file when it notices a change (like Notepad++ would).

mischnic, 2016-09-06:
The setting "Use external editor" should reload the Sketch when the Arduino window regains focus.

gitmeister, 2016-09-06:
I tried this but the IDE will proceed to "lock-out" edits done in the IDE. It may not be great practice to have both the IDE and an external editor accessing the same file but the dialog indicating that "An external program has changed this file do you want to reload" is supported by the various OS's.

per1234, 2017-03-10:
Just a note that there is another issue for the sketch modified notification in the IDE feature request, which is related but not a duplicate of this issue:
https://github.com/arduino/Arduino/issues/4551