arduino/Arduino #5346: Compiler optimization fails to generate code for code with section attribute.

gitmeister, 2016-09-05:
Removing -Os compiler option allows code to be generated.

See https://github.com/Optiboot/optiboot/issues/187

facchinm, 2016-09-06:
Hi @gitmeister ,
can you post a sketch showing the described behaviour? Variables accessed only in IRQs should be declared `volatile` and maybe `__attribute__((used))` to avoid being stripped. 
Could you also activate verbose compilation with All warnings and paste it here? Thanks!

gitmeister, 2016-09-06:
The code isn't running in an ISR, it's just placed in ".init0" so that it's the first thing running after the bootloader (which in this case is Optiboot 6.2). The code is just the "test_reset.ino" from the Optiboot project. If compiled as-is the function resetFlagsInit() will not be linked into ".init0" so the reset cause will never be available in register R2 (as provided by the bootloader). You will always see a Reset Value of 0x0. Removing the dummy line (which never executes because the reset flags will never be 0xEE) forced the code to be linked and now you'll get Reset Value 0x8 which is correct. You get the same effect if you remove -Os (Optimization).

/*
 * test_reset
 * May 2015 by Bill Westfield (WestfW)
 * Released to the public domain.
 *
 * This sketch demonstrates retrival of the Reset Cause register (MCUSR) of the AVR.
 * Normally, MCUSR itself is destroyed by the use of a bootloader, but Optiboot v4.6
 * and later save the contents in register r2, where it can be accessed by an
 * application.
 */
#include <avr/wdt.h>

/*
 * First, we need a variable to hold the reset cause that can be written before
 * early sketch initialization (that might change r2), and won't be reset by the
 * various initialization code.
 * avr-gcc provides for this via the ".noinit" section.
 */
uint8_t resetFlags __attribute__ ((section(".noinit")));

/*
 * Next, we need to put some code to save reset cause from the bootload (in r2)
 * to the variable.  Again, avr-gcc provides special code sections for this.
 */
void resetFlagsInit(void) __attribute__ ((naked)) __attribute__ ((section (".init0")));
void resetFlagsInit(void) {
  /*
   * save the reset flags passed from the bootloader
   * This is a "simple" matter of storing (STS) r2 in the special variable
   * that we have created.  We use assembler to access the right variable.
   */
  __asm__ __volatile__ ("sts %0, r2\n" : "=m" (resetFlags) :);
}

void setup() {
  Serial.begin(9600);  // Initialize serial port
  
  Serial.println("Reset flag test");
  
  Serial.print("Have reset flag value 0x");
  Serial.print(resetFlags, HEX);

  // REMOVE TO ALLOW COMPILATION
  //if (resetFlags==0xEE) resetFlagsInit();
  
  /*
   * check for the usual bits.  Note that the symnbols defined in wdt.h are
   * bit numbers, so they have to be shifted before comparison.
   */
  if (resetFlags & (1<<WDRF))
  {
    Serial.print(" Watchdog");
    resetFlags &= ~(1<<WDRF);
  }
  if (resetFlags & (1<<BORF))
  {
    Serial.print(" Brownout");
    resetFlags &= ~(1<<BORF);
  }
  if (resetFlags & (1<<EXTRF))
  {
    Serial.print(" External");
    resetFlags &= ~(1<<EXTRF);
  }
  if (resetFlags & (1<<PORF))
  {
    Serial.print(" PowerOn");
    resetFlags &= ~(1<<PORF);
[... 63 lines cut ...]
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\wiring_shift.c.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\CDC.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\HardwareSerial.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\HardwareSerial0.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\HardwareSerial1.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\HardwareSerial2.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\HardwareSerial3.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\IPAddress.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\PluggableUSB.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\Print.cpp.o"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-gcc-ar" rcs  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\core\core.a" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp\co [... cut ...] cf7bf627b121d.tmp/FIO-WDT.ino.eep"
"C:\Program Files\Arduino\hardware\tools\avr/bin/avr-objcopy" -O ihex -R .eeprom  "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp/FIO-WDT.ino.elf" "C:\DOCUME~1\Marc\LOCALS~1\Temp\build29fe7f9f7333c3f1b45cf7bf627b121d.tmp/FIO-WDT.ino.hex"

Sketch uses 1,876 bytes (5%) of program storage space. Maximum is 32,256 bytes.
Global variables use 327 bytes (15%) of dynamic memory, leaving 1,721 bytes for local variables. Maximum is 2,048 bytes.

facchinm, 2016-09-06:
Got it.
Adding `__attribute__ ((used))` to the function signature does the trick indeed (I only checked with readelf, please check functionality).
I don't know if we can classify it as a bug or a "feature", but I'd expect (as you did) that a function declaring itself in a `.init` section should not be pruned. 
Maybe escalating this to mainline is the right way to proceed. In the meantime, please test the "workaround", thanks!

matthijskooijman, 2016-09-06:
If this is indeed a bug, it's a bug in gcc. However, I suspect the main cause here is one of ordering and separated responsibility. I suspect the LTO optimizer / code generater does not see any reason why this function should be present in the binary. It does not really care about these section attributes, since that just decides *where* a function is stored. Even the linker does not really care about these functions, it just places them in the binary based on its section. I think the linker doesn't really know that these "init" functions are automatically called, that's something that the avr-libc initialization routines and linker scripts arrange for.

The reason this was not an issue before, I suspect, is that without LTO, there is no pruning of unused functions. The linker *does* do pruning of unused sections, and Arduino compiles with options to put each function into its own section, which effectively results in pruning of unused functions. However, the .init sections are referenced by the avr-libc initialization code, so these are not removed.

Summarizing: previously, the linker did both the pruning of functions and recognize the init sections, so everything worked. Now, the LTO-optimizer (i.e. the compiler part of the linking step) does the pruning, and the actual linker knows ab out sections, so there is some missing information there.

Adding the "used" attribute solves this by providing more information to the LTO-compiler. To properly fix this, I guess this needs some involvement of the gcc linker plugin or similar, but I'm not entirely sure. I'd invite you to report this case to the gcc bugzilla (search for existing reports first, and best to reduce your example further), to see what they think.

gitmeister, 2016-09-06:
1) Adding `__attribute__ ((used))` indeed allows full functionality. Thanks facchinm. 
2) What matthijskooijman says makes tons of sense and as he states the .initN sections are exceptions as they have no external linkage as they are called sequentially by avr-libc. The fact that they created the ((used)) attribute likely means they foresaw this case. I might try one last thing and try modifying the linker options to "keep" sections like ".init1". Just in case it offers an Arduino-specific solution.

gitmeister, 2016-09-06:
Ok to close if you wish.

matthijskooijman, 2016-09-06:
One more observation: ISRs are essentially similar, in that the avr-libc init code / linker scripts arrange for them to be called, but the compiler does not know about this. Hence, the `ISR` macro also adds the `used` attribute IIRC.

> EDIT: Ok, scratch that, the linker scripts already KEEP the init sections it's really an optimization thing. 

That would be consistent with my earlier conclusion about LTO being the one that removes the functions - the linker never sees them.

I'm closing this, thanks for forwarding to gcc (if you create a bug report, please post the link here for future reference).

closed 2016-09-07 by cmaglie