arduino/Arduino #5348: WiFi shield upgrade documentation

svrooij, 2016-09-06:
The [WiFiShield Upgrade Firmware Page](https://www.arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading) specifically says it should not be connected to an Arduino, when upgrading the firmware. But on my mac it refuses to upgrade the firmware while it is not connected to an Arduino.

If I plug it in to an Arduino Uno, the upgrading process works immediately.

mahanthgouda, 2018-06-04:
WiFi Sheild Upgrade documentation
On Web page : https://www.arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading

The section "Download updated firmware for the 32UC3A1256/AT32UC3A1256"
The latest version is here. Choose WiFi shield firmware.

When I click "here" the below link shows LINK 404 error:
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares

Any help would be highly appreciated.
I'm trying to upgrade the firmware on my WiFi shield.

Thanks,
Mahanth

facchinm, 2018-06-04:
Hi @mahanthgouda ,
the new url is https://github.com/arduino/ArduinoCore-avr/tree/master/firmwares/wifishield, I just updated the link on the webpage.

mahanthgouda, 2018-06-04:
Thank you for the quick response, will try this out and if any more issues, will get back to you. 

Thanks

per1234, 2018-10-27:
No, the Arduino WiFi Shield is completely different hardware from your ESP8266-based shield. Neither the firmware or installation instructions discusses here would be any use for your shield. If you haven't already done so, you can post on the forum and we'll help you out over there.

MacquarieP, 2018-10-27:
Oh.  Excellent to know thanks for that.

shinmai, 2019-05-28:
> I just updated the link on the webpage.
In the article titled [Upgrading the WiFi shield firmware](https://www.arduino.cc/en/Hacking/WiFiShieldFirmwareUpgrading) under the heading **Download updated firmware for the 32UC3A1256/AT32UC3A1256** there's a link that still points to the old 404 URL.

svrooij, 2019-05-31:
This issue is no langer relevant

closed 2019-05-31 by svrooij