arduino/Arduino #5350: bitSet, bitClear & bitWrite do not work for long long data type

Xriva2002, 2016-09-07:
The bitSet, bitClear & bitWrite macros defined in Arduino.h do not support the long long (uint64_t) data type though bitRead is fine. The problem is simply fixed by changing the bitSet & bitClear macros from...
`#define bitSet(value, bit) ((value) |= (1UL << (bit)))
#define bitClear(value, bit) ((value) &= ~(1UL << (bit)))
`
to 
`#define bitSet(value, bit) ((value) |= (1ULL << (bit)))
#define bitClear(value, bit) ((value) &= ~(1ULL << (bit)))
`

peabotai, 2016-09-07:
Much as I would like to left-shift a value of multiple millions of bits, that is not the problem.

#define bitSet(value, bit) ((value) |= (1ULL << (bit)))
#define bitClear(value, bit) ((value) &= ~(1ULL << (bit))) 

Instead:

#define bitSet(value, bit) ((value) |= (1 << (uint8_t) (bit)))
#define bitClear(value, bit) ((value) &= ~(1 << (uint8_t) bit)))
or maybe
#define bitSet(value, bit) ((value) |= (1 << (0x3F & bit)))
#define bitClear(value, bit) ((value) &= ~(1 << (0x3F & bit)))
(Untested.)

I don't actually support the idea that any change is needed.  Don't expect to shift a million bits.
Peter Olson

PaulStoffregen, 2016-09-07:
This may be stating the overly obvious, but you're requesting a change to very mature, very widely used code.

This probably ought to be common sense, but I'll say it anyway... successful projects like Arduino see tremendous risk in changing such widely used functions.

Again, perhaps common sense, any proposal for such changes needs to overcome quite a lot of concern the change could break existing code, or add inefficient results.  If you want anyone to really take this seriously, you're probably going to need to do quite a bit more verification.  At the very least, check if it increases code size or slows programs, on AVR, SAM and SAMD architecture.

PaoloP74, 2016-09-08:
same of https://github.com/arduino/Arduino/issues/1670?

Xriva2002, 2016-09-08:
@peabotai thanks for your suggestion(s) but they do not work and even degrade the functions so they only work on 8 bit data types where the old ones worked on unsigned long (usually uint32_t) data types

> This probably ought to be common sense, but I'll say it anyway... successful projects like Arduino see tremendous risk in changing such widely used functions.
Then I'm surprised it has never been mentioned/spotted before this. After doing a grep search of the entire Arduino 1.6.9 folder structure for bitSet/bitClear/bitWrite I would also dispute the "widely used functions" claim

@PaoloP74 Well spotted, I had not seen that. I will close this Issue as it has already been reported.

closed 2016-09-08 by Xriva2002