arduino/Arduino #5351: 1.6.11 will not start

mattpackwood, 2016-09-07:
I am unable to start 1.6.11 on either my Mac or PC, I get the splash screen for about 2 seconds then nothing....

facchinm, 2016-09-07:
Hi @mattpackwood , could you be more specific?
Please send the output you'll obtain by launching `arduino-debug.exe` (on Win) or  `/Applications/Arduino.app/Contents/MacOS/Arduino` from OSX Terminal, thanks

mattpackwood, 2016-09-08:
I get the following….

Mac:
java.lang.NullPointerException
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:123)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:614)
	at processing.app.Base.<init>(Base.java:296)
	at processing.app.Base.guardedMain(Base.java:223)
	at processing.app.Base.main(Base.java:136)

PC
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(Con
tributionsIndexer.java:123)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:614)
        at processing.app.Base.<init>(Base.java:296)
        at processing.app.Base.guardedMain(Base.java:223)
        at processing.app.Base.main(Base.java:136)

per1234, 2016-09-08:
As I suspected that's the same error output reported at http://forum.arduino.cc/index.php?topic=419071.msg2885938#msg2885938 which was caused by an incorrectly formatted 3rd party hardware JSON file(missing `tools` field) in that case. I pointed that out to you yesterday on your forum post(https://forum.arduino.cc/index.php?topic=422667) but you never bothered to respond.

Please don't cross post, or at least be considerate enough to include links to your other posts.

mattpackwood, 2016-09-08:
Sorry about that, my spam filter trapped the forum response.  Was not ignoring you, message was trapped.

Did not know the best communications method to use so hit both, for future reference which is better Forum or GitHub?

Matt

mattpackwood, 2016-09-08:
By the way, I se instructions on how to fix for PC but not Mac, do you know where the path is not he config folder I need to remove?

per1234, 2016-09-09:
>for future reference which is better Forum or GitHub?

Forum for support, GitHub to report bugs. In this case it would be best to try the forum first to see if we could get the issue resolved or if a bug report needed to be submitted it could have more information. You could probably consider the issue reported at http://forum.arduino.cc/index.php?topic=419071.msg2885938#msg2885938 to be a bug since this formatting was specifically allowed for in all previous versions after https://github.com/arduino/Arduino/commit/1b7574a8b9ac5cbbac4809bdd0a6a27014d7ea26. This was done because the lack of a `tools` field was causing different problems at that time. I decided that since it was easily fixed by the 3rd party hardware developers and only affected 3rd party hardware packages that it was best not to bother the Arduino developers. Instead I submitted pull requests to all the offending packages I found on the (Unofficial list of 3rd party boards support urls)[https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls] list(and actually I had fixed this issue in a bunch of repositories back when the original issue occurred).

>do you know where the path is not he config folder I need to remove?

/Users/{username}/Library/Arduino15/preferences.txt
It might be a hidden folder.

If that solves your problem please let me know what the offending URL is and I'll check if it still needs to be fixed.

matthijskooijman, 2016-09-09:
FWIW, this is also a bug in the IDE itself that needs to be fixed. Instead of silently failing, it should show an error message with the offending url, skip it and continue. I'll change the title of this report accordingly.

mattpackwood, 2016-09-09:
> Forum for support, GitHub to report bugs. In this case it would be best to try the forum first to see if we could get the issue resolved or if a bug report needed to be submitted it could have more information. You could probably consider the issue reported at http://forum.arduino.cc/index.php?topic=419071.msg2885938#msg2885938 to be a bug since this formatting was specifically allowed for in all previous versions after 1b7574a. This was done because the lack of a tools field was causing different problems at that time. I decided that since it was easily fixed by the 3rd party hardware developers and only affected 3rd party hardware packages that it was best not to bother the Arduino developers. Instead I submitted pull requests to all the offending packages I found on the (Unofficial list of 3rd party boards support urls)[https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls] list(and actually I had fixed this issue in a bunch of repositories back when the original issue occurred).

I was not able to identify which URL it was but this is the list of all the ones I was using...

Arduino Boards:

https://adafruit.github.io/arduino-board-index/package_adafruit_avr_index.json
https://raw.githubusercontent.com/EagleWorks/hexbright/master/package_hexbright_index.json
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
https://raw.githubusercontent.com/EagleWorks/hexbright/master/package_hexbright_index.json
https://cdn.rawgit.com/geolink/opentracker-arduino-board/master/package_opentracker_index.json
https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json
http://rfduino.com/package_rfduino_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json

per1234, 2016-09-09:
The problem URL was
https://raw.githubusercontent.com/EagleWorks/hexbright/master/package_hexbright_index.json
which was missing the `tools` field. I submitted a pull request to fix the issue and it's already been merged. I've now tested Arduino IDE 1.6.11 with all those URLs without problems.

matthijskooijman, 2016-09-09:
@per1234, ah, awesome! You probably explained this already in the forum link I didn't read? In any case, I think this issue can then be closed.

matthijskooijman, 2016-09-09:
Oh wait, I misread: You fixed the issue on the JSON side, but the IDE should really not crash on it. Reopening :-)

per1234, 2018-11-10:
@mattpackwood we should leave this issue open until it's either fixed or declared wontfix by the Arduino developers. Although it doesn't happen so often anymore now that the bulk of the 3rd party JSON files have been fixed, the ability for any malformed JSON file to completely break the Arduino IDE is certainly a bug.

mattpackwood, 2024-01-08:
Tidying up as the mechanism is significantly different in the new architecture....

closed 2024-01-08 by mattpackwood