arduino/Arduino #5352: ERROR STM32F103: Failed to init device. stm32flash Arduino_STM32_0.9 - Mac OS X El Capitan

RagnarokV, 2016-09-07:
Hey everyone.

I got my hands on an STM32F103C8t6 mini development board. I want to use it using Arduino IDE (I use a Mac, currently OS X El Capitan).

I download the zip files for STM32 in Arduino and followed Roger Clark's procedures. I have also installed Arduino SAM from Boards Manager.

1) When I connect my STM using FTDI RS232, my com port remains **/dev/cu.bluetooth-incoming-port**. What should I do?

2) When I try compiling the default Blink program, I get this error:
**Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

Sketch uses 7,060 bytes (10%) of program storage space. Maximum is 65,536 bytes.
Global variables use 1,968 bytes of dynamic memory.
Failed to init device.
stm32flash Arduino_STM32_0.9

http://github.com/rogerclarkmelbourne/arduino_stm32

Using Parser : Raw BINARY
Interface serial_posix: 230400 8E1**

What am I supposed to do/try here?

PS: Roger Clark says  to place the STM_32 folder into Arduino/hardware. I do not have this folder over here. For me, it is at Arduino/Contents/Java/hardware. Should I place it here? Or create a hardware folder at Arduino? 

PPS: As you may have realized, I am a complete newbie! Any help/advice is appreciated! =D

per1234, 2016-09-07:
This is the wrong place to report this issue. You're having a problem with arduino_stm32, the Arduino IDE developers have nothing to do with that project. If you had taken the time to check the http://github.com/rogerclarkmelbourne/arduino_stm32 page you would see there is a [wiki](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki) and a support forum: http://www.stm32duino.com. Spend some time researching and if you still don't find a solution then ask on that forum.

matthijskooijman, 2016-09-07:
As @per1234  says, I'm closing this issue.

closed 2016-09-07 by matthijskooijman