arduino/Arduino #5353: Library manager descriptions unreadable

FabianInostroza, 2016-09-07:
When using the library manager and a dark desktop theme the descriptions off the libraries are unreadable. It seems like somebody forgot to set the font color.

See screenshot
[image]

PD: I'm using numix gtk theme

q2dg, 2016-09-08:
"Liberías" instead of "Librerías"?? I see a typo there, too.

FabianInostroza, 2016-09-08:
@q2dg You are right!, I didn't notice that typo.

per1234, 2016-09-08:
> "Liberías" instead of "Librerías"?? I see a typo there, too.

Looks like there are two instances of that:
https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/i18n/Resources_es.po
and
https://github.com/arduino/Arduino/blob/master/arduino-core/src/processing/app/i18n/Resources_es.po

sdale28, 2016-09-09:
Hi there. I fixed the spelling issues in this pull request: https://github.com/arduino/Arduino/pull/5357
Though I'm not sure how to do the color scheme issue. Cheers!

cousteaulecommandant, 2016-09-09:
...plus I'd rather call them "bibliotecas" (although "librerías" seems to be accepted too).

sdale28, 2016-09-09:
I agree it seems like "bibliotecas" is more correct, but I'm not a native speaker.

FabianInostroza, 2016-09-10:
I'm a native spanish speaker (chilean) and I think that "librerías" is more used (at least in Chile).

q2dg, 2016-09-10:
Well, I agree...I'm spanish speaker (from Spain) and although "bibliotecas" would be the "correct" translation, "librerías" is much more used instead.

closed 2016-11-25 by cmaglie

FabianInostroza, 2017-03-09:
The same problem exists in the board manager

PR #5359 (2016-11-25): Set default color of text for descriptions shown in library manager.
Should fix #5353