arduino/Arduino #5358: Making Intel Edison Board Configuration available through LTSP Server

vikas1369, 2016-09-10:
Hi,
In our college we have LTSP server on which we have all the software installed and these software are made available to the student through network boot on LTSP Client. The issue I am facing is this
1. After installing Arduino IDE on LTSP server, I can't access it's GUI so I can't install the board configuration by going into Board Manager
2. Since on the server Arduino is installed without Intel Edison, when accessing Arduino IDE on client side we have to install the Board Configuration for Intel Edison manually 
3. After manually installing Intel Edison Board Configuration on client side, If I try to compile a program by selecting the board and port I get the following compilation error

fork/exec /home/admuser6/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0/i686/sysroots/pokysdk/usr/bin/i586-poky-linux/i586-poky-linux-g++: no such file or directory
Error compiling for board Intel® Edison

So, my question is-> Is there any way we can make Intel Edison Board configuration available in the Arduino IDE which is installed on the server? or What could be the way to solve this problem so that all the students are able to work on Arduino IDE and Intel Edison.

Thanks,

facchinm, 2016-09-12:
To be able to work completely on the server you probably need to transform your installation into a "portable" one (instructions here http://www.instructables.com/id/Portable-installation-guide-of-Arduino-IDE-v165-fo/?ALLSTEPS ). In this way, all files are contained in the installation directory and the user doesn't need any particular operation (make sure that the toolchain have the right executable bits set)
Them, you need to access the thin client's serial ports, so you need to execute the IDE using LocalApp (https://wiki.ubuntu.com/LTSPLocalAppSetup)
Let us know if it works correctly :wink:

sandeepmistry, 2017-02-16:
Closing this for now due to lack of activity.

closed 2017-02-16 by sandeepmistry