arduino/Arduino #5360: Cannot compile larger projects on macOS Sierra GM

mperdue, 2016-09-11:
I cannot compile the larger projects that I have on macOS Sierra but smaller sketches compile without problems. Two projects I've worked with that show the problem are Marlin and Repetier 3D printer firmware.

https://github.com/MarlinFirmware/Marlin
https://github.com/repetier/Repetier-Firmware

I've tried every IDE version from 1.6.9 to the 1.6.11 with no difference in results.

=====Error dump======

Arduino: 1.6.11 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

failed MSpanList_Insert 0x3205e0 0xfff2626f82e 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2cadd0, 0x3205e0)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Alloc(0x2cad60, 0x3, 0x1000000003f, 0xf989)
	/usr/local/go/src/runtime/mheap.c:240 +0x66
runtime.MCentral_CacheSpan(0x2d5d78, 0xc2083854c0)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167
runtime.MCache_Refill(0x30c000, 0x3f, 0x1808012000)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0

goroutine 1 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc2083852f0 sp=0xc2083852e8
runtime.mallocgc(0x6000, 0x158fe0, 0xc200000001, 0x32950)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0xc2083853a0 sp=0xc2083852f0
runtime.newarray(0x158fe0, 0x5de1, 0x1af160)
	/usr/local/go/src/runtime/malloc.go:365 +0xc1 fp=0xc2083853d8 sp=0xc2083853a0
runtime.makeslice(0x150be0, 0x0, 0x5de1, 0x0, 0x0, 0x0)
	/usr/local/go/src/runtime/slice.go:32 +0x15c fp=0xc208385420 sp=0xc2083853d8
io/ioutil.readAll(0x31e3a0, 0xc208040428, 0x5de1, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/io/ioutil/ioutil.go:19 +0xc0 fp=0xc2083854c0 sp=0xc208385420
io/ioutil.ReadFile(0xc2080d7e00, 0x4c, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/io/ioutil/ioutil.go:70 +0x1b5 fp=0xc208385588 sp=0xc2083854c0
arduino.cc/builder.(*AdditionalSketchFilesCopier).Run(0x2c49b0, 0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/additional_sketch_files_copier.go:66 +0x39f fp=0xc208385718 sp=0xc208385588
arduino.cc/builder.(*ContainerMergeCopySketchFiles).Run(0x2c49b0, 0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/container_merge_copy_sketch_files.go:48 +0x241 fp=0xc208385800 sp=0xc208385718
arduino.cc/builder.runCommands(0xc208068000, 0xc208385a98, 0x22, 0x22, 0x1, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:189 +0x139 fp=0xc208385890 sp=0xc208385800
arduino.cc/builder.(*Builder).Run(0xc208385ce0, 0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:124 +0x1227 fp=0xc208385cc0 sp=0xc208385890
arduino.cc/builder.RunBuilder(0xc208068000, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:220 +0x49 fp=0xc208385ce8 sp=0xc208385cc0
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/arduino-builder/main.go:316 +0x171b fp=0xc208385f98 sp=0xc208385ce8
runtime.main()
	/usr/local/go/src/runtime/proc.go:63 +0xf3 fp=0xc208385fe0 sp=0xc208385f98
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc208385fe8 sp=0xc208385fe0
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

emelianov, 2016-09-11:
Moreover that's enough to put .cpp/.hpp library to library folder to fail compilation of any sketch.
I put https://github.com/adafruit/TinyXML (that i'm using in another project) and try to compile Blink sample.
Compilation immediately crashes:
=== Error dump ===
failed MSpanList_Insert 0x3157a8 0x8b646240286 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2ccd50, 0x3157a8)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Free(0x2ccce0, 0x3157a8, 0xc200000000)
	/usr/local/go/src/runtime/mheap.c:500 +0x5b
runtime.MCentral_FreeSpan(0x2d4398, 0x3157a8, 0x2e, 0xc20819a000, 0xc20819bef0, 0x30e000, 0x64)
	/usr/local/go/src/runtime/mcentral.c:181 +0x1bb
runtime.MSpan_Sweep(0x3157a8, 0x4300000000, 0xc200000001)
	/usr/local/go/src/runtime/mgc0.c:1099 +0x477

goroutine 3 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc20801dfa8 sp=0xc20801dfa0
runtime.gosweepone(0x1)
	/usr/local/go/src/runtime/mgc0.c:1168 +0x1e fp=0xc20801dfc0 sp=0xc20801dfa8
runtime.bgsweep()
	/usr/local/go/src/runtime/mgc0.go:85 +0x36 fp=0xc20801dfe0 sp=0xc20801dfc0
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc20801dfe8 sp=0xc20801dfe0
created by gc
	/usr/local/go/src/runtime/mgc0.c:1386

goroutine 1 [chan receive]:
os/exec.(*Cmd).Wait(0xc20803a3c0, 0x0, 0x0)
	/usr/local/go/src/os/exec/exec.go:369 +0x30b
os/exec.(*Cmd).Run(0xc20803a3c0, 0x0, 0x0)
	/usr/local/go/src/os/exec/exec.go:246 +0x71
os/exec.(*Cmd).Output(0xc20803a3c0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/os/exec/exec.go:392 +0x1d3
arduino.cc/builder/ctags.(*CTagsRunner).Run(0x2c69a0, 0xc208020e00, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/ctags/ctags_runner.go:67 +0x5f3
arduino.cc/builder.(*ContainerAddPrototypes).Run(0x2c69a0, 0xc208020e00, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/container_add_prototypes.go:59 +0x7a8
arduino.cc/builder.runCommands(0xc208020e00, 0xc20802ba98, 0x22, 0x22, 0x1, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:189 +0x139
arduino.cc/builder.(*Builder).Run(0xc20802bce0, 0xc208020e00, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:124 +0x1227
arduino.cc/builder.RunBuilder(0xc208020e00, 0x0, 0x0)
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/builder/builder.go:220 +0x49
main.main()
	/Users/jenkins/jenkins/workspace/arduino-builder-macosx/src/arduino.cc/arduino-builder/main.go:316 +0x171b

goroutine 7 [runnable]:
os/exec.func·004(0xc208199600)
	/usr/local/go/src/os/exec/exec.go:327
created by os/exec.(*Cmd).Start
	/usr/local/go/src/os/exec/exec.go:329 +0xb6d
/private/var/folders/n4/6yrgfj795m3947f640yq1jv0000102/T/AppTranslocation/08749EA0-56B5-42A4-9BAC-1309DF65981A/d/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Arduino Leonardo.

thinkyhead, 2016-09-12:
Seems to be a general Go language problem. Check out https://github.com/golang/go/issues/16570 and https://go-review.googlesource.com/#/c/25481/ for example.

iMikie, 2016-09-21:
In case it wasn't clear above: this is solved in IDE 1.6.12 Get it from the main Arduino download page.  It was fascinating to watch the discussion unfold.  Throw 'em a contribution while you're at it!

closed 2016-09-22 by cmaglie

Snortimer, 2016-10-23:
I'm getting this with 1.6.12, and I'm dead in the water for updating my printer firmware (and can't do a job because of it). I was successful compiling and flashing firmware until I upgraded to Sierra; no other changes were made.

I've attempted to compile in 1.6.4, 1.6 12, 1.7.1, and 1.6.7.

It says it's missing u8g.h
__

Netoperz, 2016-10-23:
Same here, everything was fine till installing new os . It's kind of go related i've heard.

Any suggestionshow to fix that and when it will be fixed, downgreading the machines is not possible and whole project is halted because of that issue as we need to test firmware.

Do I must buy windows laptop or what ?

Everytime Apple is publishing new system always arduino is not working and compilers are freaking out. Does anybody test the beta OS X for fixing the bugs before GM showing on software update ?

LOL, same problems year by year....

iMikie, 2016-10-24:
I had this problem too and **installed 1.6.12** and it's working again.  Does your IDE window title include | Arduino 1.6.12?

rccursach, 2016-10-24:
I had the same problem, just removed and installed again a third party platform and things went back to normal for 1.6.9 on macOS Sierra

Snortimer, 2016-10-25:
I got it working. I'll check when I get home and post what I did (edit 3-4 files).

-Joel Hall

pdailey, 2017-01-10:
@Snortimer did you ever find time to document the fix?

Snortimer, 2017-01-16:
Of course not! I moved on to other stuff.

I'll have to look up my fix again and figure it out. It only takes about 3 minutes to implement it.

On the other hand, if you have Windows (which I do on a VM), you can compile it there. I'm not sure what changed with the Mac that made it not work.

I'm sorry I forgot about this! It just sort of fell into the background once I had other stuff to do.

wenhuizhang, 2017-01-19:
updating the avr board from board manager in IDE to 1.6.0 helps for me :)

MacroMachines, 2017-02-14:
@Snortimer would really love to hear your fix.  This is a pretty gnarly issue and can't seem to find a fix on the net, nor do I know where to start figuring out what you may have done with the file edits.

pdailey, 2017-02-14:
@MacroMachines I ended up upgrading to the latest Arduino IDE (1.8.1) and haven't seen the error since. This worked on both mac and windows.