arduino/Arduino #5361: Should not compile with -fpermissive?

matthijskooijman, 2016-09-11:
Today, I ran into some problem, which turned out to be caused by the compiler happily passing an `int` to a function that expected a pointer. Closer investigation showed that a warning was issued, where I would have expected an error. This seems to because of the `-fpermissive` flag, which gets passed by the avr core by default. According to the gcc manpage:

       -fpermissive
           Downgrade some diagnostics about nonconformant code from
           errors to warnings.  Thus, using -fpermissive allows some
           nonconforming code to compile.

I'm not sure what changes this makes exactly and if we need it at all. If the other changes are similar to this one, I think it should be disabled at some point (but it will probably break existing code that isn't quite correct, but now happens to work)...

For reference, here's a simplified version of my problem:

```
void func(Foo *ptr);

int main() {
  func(1);
  return 0;
}
```

(the real problem was more subtle, involving a struct with a conversion operator to int, which got silently converted to a pointer containing rubbish when I forgot the `&` operator when trying to pass the struct's address to a function).

OldSteve2, 2016-09-20:
A beginner, posting on the Arduino forums, came up against this too, with the following code:-
`const char tst = "Testing all";

void setup()
{
    Serial.begin(115200);
    Serial.println(tst);
}

void loop(){}`
It compiles fine with IDE V1.6.11, but in earler versions of the IDE, it quite correctly produced the following error:-
`sketch_sep20e:1: error: invalid conversion from 'const char*' to 'char' [-fpermissive]`
I think it's a step backwards, just for the sake of making a few bad libraries compile.

rkost, 2016-10-15:
AGREE!

edgar-bonet, 2016-10-28:
This compiler flag was introduced in [commit ba0c41b](https://github.com/arduino/Arduino/commit/ba0c41b), in order to not break the build of some old libraries. I am not convinced it is a good thing to _not_ let the library authors know about the problems with their bad code.

Anyway, here is another real-world example of a user being [confused by this behavior](http://arduino.stackexchange.com/questions/30600):

> I wrote a sketch that used the `millis()` function in several places. I got odd behavior because I inadvertently used the plain `millis` in one line, without the parens. [...] Why didn't the compiler flag this as an undeclared variable?

sheerun, 2019-03-03:
This prevents mjs from compiling