arduino/Arduino #5364: [Library Manager]: add controleForno

Atzingen, 2016-09-12:
I had open an issue[1] for adding my library[2] to the library manager.
The issue was cloesed and was suposed to be done, but the library is not found in the arduino and is not avalilable in the json file[3]

[1] - The issue opened: https://github.com/arduino/Arduino/issues/5337

[2] - https://github.com/Atzingen/controleForno

[3] - The json file with all libraries information (At least is here were i thing the add was suposed to be done) http://downloads.arduino.cc/libraries/library_index.json

agdl, 2016-09-15:
Should be ok now. Let me know

Atzingen, 2016-09-15:
I can see that is added to the library .json file, but when i try to install it using the arduino it does not work (I ckick install but nothing hapens and no folder apear in documents/arduino/library).
One thing that i noted is that in the arduino library .json file, the version is set as 1.0.0, but the latest release i put in github is v2.0.1 (and also tag it with v2.0.1) as was stated in the arduino guidelines for libraryes.
I'm not very experienced with github, maybe i did somthing wrong, can you help me please ?

sandeepmistry, 2016-09-15:
Hi @Atzingen,

The current `library.properties` file has a `v` in front of the version (`v2.0.1`) in the `v2.0.1` tag. It should be just `2.0.1`.

This is not supported, I would suggest you create a new `2.0.2` release, by placing `2.0.2` in `library.properties` and then create a new `v2.0.2` tag.

Atzingen, 2016-09-15:
Ok, thankyou very much !!!
One question:
 Are the number in library.properties (2.0.2) and the tag v2.0.2 connected ? Shoul I use v2.0.2 or 2.0.2 in the tag ?

sandeepmistry, 2016-09-15:
`library.properties` must not contain a any letters, just 2.0.2.

For the tag, you can use either v2.0.2 or 2.02, it's up to you.

Atzingen, 2016-09-15:
Many Thanks !!!!

sandeepmistry, 2016-09-15:
You're welcome! Can we close this issue now?

Atzingen, 2016-09-15:
Yes ! :)

closed 2016-09-15 by Atzingen