arduino/Arduino #5370: (Bug) avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.

Nic12345, 2016-09-14:
Please help, every single program I try to upload spits out this message at me:

Arduino: 1.6.11 (Windows 10), Board: "Arduino/Genuino Uno"

Sketch uses 10,578 bytes (32%) of program storage space. Maximum is 32,256 bytes.
Global variables use 387 bytes (18%) of dynamic memory, leaving 1,661 bytes for local variables. Maximum is 2,048 bytes.
avrdude: ser_open(): can't open device "\\.\COM1": The system cannot find the file specified.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

facchinm, 2016-09-15:
Hi @Nic12345 ,
COM1 is usually a real serial port of the PC, so maybe your UNO is not getting recognized or you didn't select the correct serial port. Under "Tools/Port" menu can you see "COM1: Arduino/Genuino Uno" or is there any other entry?

Nic12345, 2016-09-17:
I see Com3

Nic12345, 2016-09-17:
Thank You so much.

I made a silly mistake!

closed 2016-09-17 by Nic12345