arduino/Arduino #5373: Wire Library issue

lettierimarco, 2016-09-15:
Hello,
I am experiencing a problem with the Wire library.
I have a raspberry that communicates i2c with Arduino Mega (2561) via i2c.
Raspberry master, Arduino slave .
when I get a read command from the raspberry to arduino and I want to read an array of 9 bytes very often, I get side raspberry 0x00 0xFF 0xFF 0xFF ... as if Arduino he could not send this array (eg wire.send (array,9);).
Instead, when I want to send to the master a string constant (eg wire.send ("hello");) this problem does not occurred.
On raspberry i'm using WiringPi for java.
there's some difference between send(array,num) and send(constant char)?
on your experience this could be related to some hardware / software wrong setup ?

sandeepmistry, 2016-09-15:
Hi @lettierimarco, please share the entire sketch you are using on the Uno.

lettierimarco, 2016-09-15:
#include <FreqMeasure.h>
#include <IRremote.h>
#include <IRremoteInt.h>
#define TWI_BUFFER_LENGTH ((RAWBUF*2)+4) 
#define RAWBUF_BYTE  TWI_BUFFER_LENGTH 
#include <Wire.h>
#include <avr/wdt.h>

byte bufferToSend [RAWBUF_BYTE];
unsigned int payloadLen=0;
void setup() {
  cli();  
  wdt_enable(WDTO_8S); 
  sei();
  wdt_reset();
  Serial.begin(115200);
  DDRC = 0b11111111; //LED ports configuration PC0 to PC7 as OUTPUT
  DDRB = DDRB | 0b01100000; //pb6 and pb5 as output
  DDRE = DDRE | 0b01000000; //pe6 as output
  PORTB = PORTB | 0b01000000; //pb6 high - PIR enable
  DDRD &= ~(1 << PD2); // set pd2 as input
  DDRD &= ~(1 << PD4); // set pd4 as input
  irrecv.enableIRIn();
  FreqMeasure.begin();
  Wire.begin(9); 
  setI2CRequestManager(requestManager);
  bufferToSend[0] = (byte)0;
  bufferToSend[1] = (byte)0;
  bufferToSend[2] = (byte) -1;
  bufferToSend[3] = (byte)-1;
  bufferToSend[4] = (byte) -1;
  bufferToSend[5] = (byte) -1;
  bufferToSend[6] = (byte) -1;
  bufferToSend[7] = (byte) -1;
  bufferToSend[8] = (byte) -1;
  payloadLen = 9;
  wdt_reset();
}
void requestManager() {
      Wire.write(bufferToSend, payloadLen);
      wdt_reset();
}

sandeepmistry, 2016-09-15:
Things seem fine to be, I just tested the following (based on the Wire [slave_sender.ino](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/Wire/examples/slave_sender/slave_sender.ino) example):

```arduino
#include <Wire.h>

void setup() {
  Wire.begin(9);
  Wire.onRequest(requestEvent);
}

void loop() {
}

void requestEvent() {
  byte response[] = { 0, 1, 2, 3, 4, 5, 6, 7, 8 };
  
  Wire.write(response, sizeof(response));
}
```

With a CHIP (Linux board), I was able to read `00 01 02 03 04 05 06 07 08` over I2C.

Can you please try this sketch?

sandeepmistry, 2016-12-12:
Closing this for now, due to lack of feedback.

closed 2016-12-12 by sandeepmistry