arduino/Arduino #5379: Packaging arduino with upstream avr-gcc or avrdude question

NicoHood, 2016-09-17:
Hey guys,
I recently got a TU for archlinux which means I can now package arduino for archlinux. The question came up if archlinux should link `avr-gcc` and/or `avrdude` with the distribution/upstream packages.

I'd like to present my opinion on this, but also would like to hear other opinions, as I might be wrong. It would be nice if one of you guys could answer, as I think you should know it best: @cmaglie @facchinm @sandeepmistry @matthijskooijman @focalintend @kriegsman

My opinion is to **not** package the upstream tools.
* Compilation size differs
* Assemler code differs, [makes it hard to debug](https://github.com/arduino/Arduino/issues/5236)
* Compiler specific code as in fastled could break
* New avrlibc definitions might give problems
* [Reverting of avrdude](https://github.com/arduino/Arduino/pull/5208) already showed such problems
* Atmel also uses 4.9.x which is quite new if you ask me

As a side note: We currently package avrdude 6.3 because of a [dependency issue](https://github.com/arduino/Arduino/issues/5094), but that is only temporary.

matthijskooijman, 2016-09-18:
@NicoHood, your arguments sound sane to me. However, does Arch have any policy on this? I know that Debian has a policy of only allowing a single piece of software to be present in one package, if other software needs it, the other software should use a dependency (I think the rationale is mostly for security, this prevents missing a copy of some vulnerable software hidden inside another package).

plantarum, 2016-09-18:
Is it even possible to package Arduino for redistribution? Debian has been stuck on version 1.05 for ages due to unresolved licensing issues: https://github.com/arduino/Arduino/pull/2703

NicoHood, 2016-10-03:
So I've still been trying to package arduino with the upstream avr-gcc. I [had to solve some issues](https://github.com/arduino/Arduino/issues/660#issuecomment-251142922) first, now the packaging can continue.

Its a good practise to download all files via makepkg and not within the build script itself. Its quite a lot to download and then i found the `light-bundle` option. I've tried it with `ant clean build -Dlight_bundle=true`, but that gave me some errors when launching the IDE in the 2nd run:

```
java.io.FileNotFoundException: /usr/share/arduino/hardware/package_index_bundled.json (No such file or directory)
	at java.io.FileInputStream.open0(Native Method)
	at java.io.FileInputStream.open(FileInputStream.java:195)
	at java.io.FileInputStream.<init>(FileInputStream.java:138)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:194)
	at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:86)
	at processing.app.BaseNoGui.initPackages(BaseNoGui.java:612)
	at processing.app.Base.<init>(Base.java:292)
	at processing.app.Base.guardedMain(Base.java:219)
	at processing.app.Base.main(Base.java:132)
```

If I see it correct the light version does not download:
* avr-gcc nor avrdude
* no builtin libraries
* no avr-core

If I get it right this could all be downloadable via an arduino avr-core update, isnt it?

I don't really know for what this is used, but it could be handy for packing arduino for several distributions. The libraries are not essentially required if you ask me. You can redownload those via the boards manager. Most of the time they are out of date anyways. The avrcore can also be redownloaded, as the sam core is also not included by default. I am not sure if this would then also pull in the bundled avr-gcc or not, but it would be cool to use the system avr-gcc if its set by default.

So can anyone help me to explain this feature a bit more?

Salamandar, 2016-10-31:
Hi there !
My position can be ambiguious. It'd be great to use the upstream `avr-gcc, avr-libc, avrdude`. That would mean a lighter Arduino package, and, of course, simpler management + understanding of the Arduino toolchain ("uh, why do I need this GCC version ?").

But… Well I'm on Archlinux, and the Arduino package is being splitted. And Marlin (for instance) does not work anymore : It compiles but does not seem to execute.

I downloaded the Arduino IDE, and using the `arduino-builder` and `avrdude` from community BUT avr-gcc libc from the Arduino IDE, that works. 

`/usr/share/arduino/hardware/archlinux-arduino/avr/platform.txt` : 
```
- compiler.path=/usr/bin/
+ compiler.path=/home/salamandar/Téléchargements/arduino-1.6.12/hardware/tools/avr/bin/
```

Maybe `avr-gcc6` is incompatible with the arduino hardware libs ? Maybe you should provide a Gcc4.9 package ?

NicoHood, 2016-10-31:
@Salamandar It is still possible. See: https://wiki.archlinux.org/index.php/arduino#AVR_Boards