arduino/Arduino #5380: Edit > Find issues

per1234, 2016-09-17:
Using Arduino IDE 1.6.12 Hourly Build 2016/09/14 08:20 with Windows 7 64 bit
- Create a sketch with 3 or more tabs.
- Add the text **findtext**(or any text you like) to each tab.
- Save the sketch.
- Close the sketch.
- Open the sketch.
- **Edit > Find > Find: > findtext**
- Check the **Search all sketch tabs** box

**Issue 1:**
- Click **Find** button repeatedly -The found text on all but the first tab is not highlighted. If you click on the IDE window(not on the editor area as this will cause it to be deselected) the highlight will appear. Once that has been done, the highlight will appear on that tab on subsequent finds. This is the reason the sketch had to be closed and reopened, because the process of creating the tabs/adding the **findtext** to each tab causes the highlighting to work correctly.

**Issue 2:**
-  Find goes through the tabs right to left instead of the previous left to right behavior.

**Issue 3:**
- **Replace with: > replaced**
- Click **Replace all**
- The **findtext** in the second tab was not replaced.

These issues don't occur with Arduino IDE 1.6.11 or previous

Please don't release Arduino IDE 1.6.12 before this issue is fixed. It's very important for **Replace All** to work reliably.

matthijskooijman, 2016-09-18:
This is probably caused by #4363. I'll see if I can find some time to look into this.

facchinm, 2016-09-19:
Should be fixed by #5383
#per1234, could you check as soon as @ArduinoBot prepares the builds?

matthijskooijman, 2016-09-19:
I just came to the same conclusion. However, it seems this only fixes issue 1 and 3, I still see 2 happening. What seems to happen there is that when a tab is first created, it does not have a caret yet (or something like that) and selections do not show up. Once you click in the text area, selections will show up from then on. I'll see if I can figure out what happens exactly.

facchinm, 2016-09-19:
Issue 2 looks ok to me too (have I missed something about the real problem? )

matthijskooijman, 2016-09-19:
@facchinm Could you reproduce the problem without the fix? Note that the issue only occurs in tab that have never received the focus, so really only directly after loading a sketch.

facchinm, 2016-09-19:
Ah ok, we are talking about issue 1 (find does not highlight content on secondary tabs).
Indeed, it's still here after the patch...

matthijskooijman, 2016-09-19:
Oh, woops, I meant issue 2/3 is fixed, 1 is still present. Sorry for that.

It seems the issue is related to focus somehow, I think that only after the textarea has been focused at least once, it draws its selection or something like that. The problem is probably that the selection is set while the editor window is not focused. Not sure how it works exactly yet, though.

PR #5383 (2016-09-20): Fix nextTab/prevTab selection logic
fixes #5380
commit 1efa07f0 (2016-09-20): Enable visibility of selection on find

Fix #5380
closed 2016-09-20 by facchinm