arduino/Arduino #5393: [website] Hourly build date shown on download page doesn't match the date shown on the IDE's window title

per1234, 2016-09-21:
https://www.arduino.cc/en/Main/Software#hourly currently shows:
>LAST UPDATE
8 September 2016 13:36:32 GMT

but when I download the hourly build the window title says:
>2016/09/20 07:05

This can cause unnecessary large downloads, wasting bandwidth, because that's the only way(at least that I know of) to find if there is a newer version than the one I already have installed available.

Chris--A, 2016-09-21:
You can always check the date of the last commit to master. If it is a few hours old, there is probably a new hourly build available.

mastrolinux, 2016-09-21:
To display the date in the web page we use the https://downloads.arduino.cc/hourly_dates.json values. @cmaglie  do you know the reason of the discrepancy?

mastrolinux, 2016-10-04:
This one has been solved.

closed 2016-10-04 by mastrolinux