arduino/Arduino #5394: CRC mismatch of downloaded file causes all subsequent Boards Manager installation attempts of that package to fail

per1234, 2016-09-21:
Using Arduino IDE 1.6.12 Hourly Build 2016/09/20 07:05 with Windows 7 64 bit

If a file downloaded during a Boards Manager installation has a CRC mismatch that file will cause attempts to reinstall to fail because Boards Manager doesn't redownload files that already exist in the `staging` folder. This could also happen if a previously installed package used the same filename for a file with a different hash. The solution would be to delete the corrupted file after CRC mismatch is encountered. This would still cause the installation to fail but will not block later attempts. Right now the only way around the issue is to manually delete the corrupted file from the `staging` folder.

facchinm, 2016-09-21:
This is a known issue, it shouldn't happen since 
1 - all the downloads should be saved only if the crc is valid (this is not handled ATM)
2 - two different packages should never have the same name (we try our best to avoid name clashing)

It is surely something we need to tackle once we start Board Manager refactoring.

per1234, 2016-09-21:
Sorry if it's a duplicate. I did some searching and found other CRC error reports but only closed issues with instructions to manually delete the file, not discussions of how to fix the problem.

I agree it's best practices to make sure that all filenames are unique but the creators of 3rd party hardware packages may not always follow that since it's not obvious it will cause a problem. I could see things like 1.0.0.zip or common tool filenames causing this issue. Looking through the first part of the Unofficial list of 3rd party boards support urls wiki page I did find some filenames in the JSON files that could potentially cause this issue.

I believe the same issue happens with Library Manager also. I can't completely test it but if I fake it by manually corrupting one of the files in the staging/libraries folder and then trying to install that library it does cause the same issue. I can't test whether Library Manager saves downloaded files with a CRC mismatch so I could be wrong about that.

pfeerick, 2016-09-26:
Yeah, this happens with the library manager also. I just tried to installed a library in a different sketchbook, and it wouldn't install due to `CRC doesn't match. File is corrupted.` . I then went and deleted the cached file from the staging folder, and it downloaded and installed fine. I'm wondering if the case where the file already exists/has been previously downloaded isn't being handled properly? I seem to think it isn't, as I had issues with board packages a few months that were fixed by deleting all the cached packages so the IDE actually had to go and download them.

jmurphy644, 2016-09-27:
pfeerick's workaround worked for me as well, just removing everything from the staging folder.

PR #5453 (2016-10-14): If a downloaded archive is corrupted (CRC error) retry the download
Previously the CRC error was quite annoying to recover because the user needed to manually delete the corrupted file from the staging folder (without knowing the exact path of the file to remove).

Now the IDE tries autonomously to resolve the situation by removing the file and downloading it again.

Fixes #5394 #4303
commit 7883835b (2016-10-14): If an archive is corrupted (CRC error) retry the download

Previously the CRC error was quite annoying to recover because
the user needed to manually delete the corrupted file from the
staging folder (without knowing the exact path of the file to
remove).

Now the IDE tries autonomously to resolve the situation by
removing the file and downloading it again.

Fixes #5394 #4303
closed 2016-10-14 by cmaglie