arduino/Arduino #5398: Feature request: Include boards package version in "Copy error messages" output

per1234, 2016-09-21:
Hardware package/platform version is becoming increasingly important troubleshooting information. It's very useful to get all necessary information to help someone with an error by just having them click the "Copy error messages" button and paste it to the forum. Many users have no clue which platform version they are using and don't understand the distinction between the similar, but not identical, Arduino IDE version and Arduino AVR Boards version.

matthijskooijman, 2016-09-21:
Interestingly I just started this morning on adding this kind of information to the arduino-builder output (for a different purpose, but the result is the same). I'll see if I can get this working quickly.

What I'm planning is to change:

    Using board 'atmega328bb' from platform in folder: /some/path/hardware/breadboard/avr
    Using core 'arduino' from platform in folder: /some/path/hardware/arduino/avr

Into:

    Using board 'atmega328bb' from platform in folder: /some/path/hardware/breadboard/avr
    Using core 'arduino' from platform 'Arduino AVR Boards' version 1.6.13 in folder: /some/path/hardware/arduino/avr

(Note that the breadboard platform is missing a platform.txt and thus
name and version, so they are not listed, otherwise they would be).

Does that look like it would fulfill your request?

per1234, 2016-09-21:
It looks like a useful improvement but the problem is that information is only shown if **File > Preferences > Show verbose output during: > compilation** is checked. Compilation verbose output is not enabled by default and is rarely needed for troubleshooting the common issues reported on the forum. I usually have it turned off because it makes for more scrolling to read the warnings. Requesting the user to enable it adds an additional step to the support process and can make the output longer than the maximum character limit of the forum, forcing the user to add it as an attachment or dropbox link, etc. which adds yet another level of complication and makes it harder for the other forum users to view the output. Having verbose output during compilation turned off makes it less likely that the platform version can be determined from platform paths shown in the output as those paths will only be shown if the error involves one of the platform files.

I was referring the text added to the start of the output only when the "Copy error messages" button is used(not shown on the console). For example, changing:
```
Arduino: 1.6.12 (Windows 7), Board: "Arduino/Genuino Uno"
```
to
```
Arduino: 1.6.12 (Windows 7), Board: "Arduino/Genuino Uno", Platform version: 1.6.14
```
Adding the platform name to this output would also be useful.