arduino/Arduino #5399: New Library - EmonLib

ricardojlrufino, 2016-09-21:
Please add 
https://github.com/openenergymonitor/EmonLib

cmaglie, 2016-10-03:
The library is missing a `library.properties` file.
It also needs a tag pointing to a revision with the correct library.properties.

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cmaglie, 2016-10-18:
Closing this for now, please reopen when you're done.

closed 2016-10-18 by cmaglie

ricardojlrufino, 2016-10-18:
Sorry for not giving feedback.

I asked the developer put the settings, but he did not put

PS: I'm not a developer, but I find it interesting that library is the repository

glynhudson, 2016-10-18:
`library.properties` has been added, sorry for the delay. 

https://github.com/openenergymonitor/EmonLib/blob/master/library.properties
https://github.com/openenergymonitor/EmonLib/pull/31

cmaglie, 2016-11-02:
I've added the library, BTW, the tag 1.1.0 doesn't match the library.properties:

https://github.com/openenergymonitor/EmonLib/tree/1.1.0

actually the library.properties is not present at all in the tag 1.1.0, probably the tag has been made before the file has committed. Please redo the tag, and the library will be picked up by our server automatically (within an hour).

glynhudson, 2016-11-03:
Fixed. Thanks for letting me know. I've redone the tag to include
library.properties