arduino/Arduino #5404: IDE 1.6.12 - WARNING on Burn Bootloader

logicaprogrammabile, 2016-09-22:
hi 
i burned bootloader to an atmega328p with ArduinoISP and the result is:

>  ***failed;  
> avrdude: WARNING: invalid value for unused bits in fuse "efuse", should be set to 1 according to datasheet
> This behaviour is deprecated and will result in an error in future version
> You probably want to use 0xfd instead of 0x05 (double check with your datasheet first).

but the ide returns:
**done burning bootloader**

i have modified the board.txt at line
`uno.bootloader.extended_fuses=0x05`

to
`uno.bootloader.extended_fuses=0xFD`

This change eliminates the warning message
it's correct?

tnks

facchinm, 2016-09-23:
It's indeed the correct behaviour. The first run didn't fail indeed (I don't know where that `***failed;` comes from). We still didn't merge https://github.com/arduino/Arduino/pull/5182 for compatibility, we'll do it in one of the future releases.
Closing as expected (please reopen it if the first run really failed => the bootloader was not burned)

closed 2016-09-23 by facchinm

JeroenIoT, 2016-10-18:
Thanks. I had the same message / warning when burning bootloader. However the funny thing is that when I change the boards.txt (to 0xfd) I still get the same warning (restarted of course and using Arduino IDE 1.6.12)

RichieDevereux, 2021-11-08:
Where do I find the boards.txt file?