arduino/Arduino #5408: Update Serial.print() documentation

beikeland, 2016-09-23:
https://www.arduino.cc/en/Serial/Print says "As of version 1.0, serial transmission is asynchronous; Serial.print() will return before any characters are transmitted."

I think it could be beneficial to update this to include that it is only non-blocking until you exceed the transmit buffer, and the current size of the buffer, or a link to where the buffer size is defined to avoid potentially stale information both on the docs, and users finding stale information when turning to google to search for the answers.

q2dg, 2016-09-23:
I agree but maybe a little short explanation about what a buffer is should be grateful for newbies.

aknrdureegaesr, 2017-02-10:
Where is the source code for https://www.arduino.cc/en/Serial/Print , so one can make a pull request to fix this?

matthijskooijman, 2017-02-10:
@aknrdureegaesr 
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/HardwareSerial.h
https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Print.h
(and corresponding cpp files)

per1234, 2019-02-08:
Closing as fixed by https://github.com/arduino/reference-en/pull/520

closed 2019-02-08 by per1234