arduino/Arduino #5413: Command line builder error when setting the port (v1.6.12)

Erel2, 2016-09-25:
Command line example: arduino_debug.exe --upload <project>  --port COM54

Output:

Loading configuration...
java.lang.NullPointerException
        at processing.app.BaseNoGui.selectSerialPort(BaseNoGui.java:1088)
        at processing.app.helpers.CommandlineParser.parseArgumentsPhase1(CommandlineParser.java:143)
        at processing.app.Base.<init>(Base.java:273)
        at processing.app.Base.guardedMain(Base.java:219)
        at processing.app.Base.main(Base.java:132)

Seems like BaseNoGui.initPackages is not called.

facchinm, 2016-09-26:
@cmaglie probably caused by https://github.com/arduino/Arduino/commit/7d27c43ff237f89dd646953f6841c425e3a1d154

hanabanana, 2016-09-30:
Not to offend anyone, but regression like this really causes big problem for us because we rely on the CLI to compile and upload. Because of Sierra, we need to upgrade to 1.6.12 but this bug breaks our code from the beginning. 
Seems like the IDE has switched from CLI to arduino-builder and separate upload command since 1.6.6(?). Does it mean the CLI will be less likely maintained or regression like this will occur more often in the future?

facchinm, 2016-10-03:
@hanabanana, you are right, most of our CI tests switched using only `arduino-builder`. 
I'm really sorry for the inconvenience and we'll immediately add this test to the testing infrastructure. 
In the meantime, you can stick with IDE 1.6.11 only replacing the builder on Sierra with [this](http://downloads.arduino.cc/tools/arduino-builder-macosx-1.3.21_r1.tar.bz2) package, or replace `arduino-core.jar` with the attached one 
[arduino-core.zip](https://github.com/arduino/Arduino/files/505849/arduino-core.zip)

closed 2016-10-03 by facchinm

hanabanana, 2016-10-03:
@facchinm Thanks for the workaround and quick fix. We will probably wait for the next release though since downloading and replacing the patches cannot be baked into our code. Any estimate of when the next release will come out?

SB618, 2016-10-04:
Yes, this needs to be regression tested.  We have code that depends on this and has caused a big compatibility issue.  Thanks for the workaround, please let us know when this will be fixed.