arduino/Arduino #5416: Editor is not scrolled to found text on tab that was not previously opened

per1234, 2016-09-25:
Using Arduino IDE 1.6.13 Hourly Build 2016/09/23 03:33 with Windows 7 64 bit

- Create a sketch with 2 tabs
- Add text to be found in the second tab on a line that will require the editor window to be scrolled down to show the text.
- Save and close the sketch
- Open the sketch
- **Edit > Find > Find: >** enter text to be found **> Search all Sketch Tabs**(check) **> Find** - the text is found in the second tab but the editor window is not scrolled down to show the found text.

The reason that the sketch must be closed and then reopened is the issue doesn't occur once the second tab has been opened, as in https://github.com/arduino/Arduino/issues/5380.

This issue doesn't occur in Arduino IDE 1.6.12 or previous.

matthijskooijman, 2016-09-27:
> This issue doesn't occur in Arduino IDE 1.6.12 or previous.

@per1234, are you sure you mean 1.6.12 here? Or 1.6.11? Since the changes in git master / hourly build since 1.6.12 are just a handful, none of which seem related to this issue.

matthijskooijman, 2016-09-27:
I dug into this a bit more, it seems that the problem is that setting the caret position normally scrolls towards it (due to [this line](http://hg.openjdk.java.net/jdk8/jdk8/jdk/file/687fd7c7986d/src/share/classes/javax/swing/text/DefaultCaret.java#l1313) called from [here](http://hg.openjdk.java.net/jdk8/jdk8/jdk/file/687fd7c7986d/src/share/classes/javax/swing/text/DefaultCaret.java#l1287). However, this only happens when `modelToView()` (called a few lines before) returns non-null. The [implementation for modelToView that ends up being used](http://hg.openjdk.java.net/jdk8/jdk8/jdk/file/687fd7c7986d/src/share/classes/javax/swing/plaf/basic/BasicTextUI.java#l1045) only does this when `getVisibleEditorRect()` returns non-null, which apparently only happens when the component has a size. A tab that is shown for the first time probably does not have a size yet, especially because `Editor.selectTab` does not immediately make the tab visible, but uses `SwingUtilities.invokeLater`. On the other hand, `repaintNewCaret()` als runs inside `invokeLater`, but perhaps some things are even further delayed.

I haven't dug in more closely yet to see when the component size gets set exactly.

per1234, 2016-09-27:
> are you sure you mean 1.6.12 here? Or 1.6.11?

Sorry, that was a typo, I meant to write 1.6.11. I've edited my report with the correction.

ninja2B, 2018-11-20:
Hey Mr Matthijs Kooijman, I understand you're a busy man, but I'm hoping I might prompt you to fix this annoying issue. As mentioned above this bug was introduced into the Arduino IDE somewhere after release 1.6.11 and it's still there in 1.8.7. 

You know the saying: "If you need something done, give it to a busy man" :-)

TIA

matthijskooijman, 2018-11-20:
Hm, seems I assigned this to myself, so I should probably have a closer look (it did kind of fall of my radar). I've added it to my todolist again, but might still take a while...

ninja2B, 2018-11-20:
Good man, thanks. Happy to wait for a positive outcome :-)