arduino/Arduino #5417: ArduinoRobot library doesn't compile

per1234, 2016-09-26:
Using Arduino IDE 1.6.13 Hourly Build 2016/09/23 03:33 with Windows 7 64 bit

Any sketch that contains the line:
```arduino
#include <ArduinoRobot.h>
```
fails to compile with a bunch of this sort of stuff:
```
`osc' referenced in section `.text' of C:\Users\per\AppData\Local\Temp\ccvUoJUW.ltrans0.ltrans.o: defined in discarded section `.text' of libraries\Robot_Control\Squawk.cpp.o (symbol from plugin)
```
I suspect this is related to LTO, however using pre-LTO Arduino AVR Boards 1.6.11 compile still fails, even for the BareMinimum sketch:
```
C:\Program Files (x86)\arduino-nightly\portable\packages\arduino\hardware\avr\1.6.11\cores\arduino\CDC.cpp: In function 'bool CDC_Setup(USBSetup&)':
C:\Program Files (x86)\arduino-nightly\portable\packages\arduino\hardware\avr\1.6.11\cores\arduino\CDC.cpp:104:29: error: 'MAGIC_KEY_POS' was not declared in this scope
    uint16_t magic_key_pos = MAGIC_KEY_POS;
                             ^
C:\Program Files (x86)\arduino-nightly\portable\packages\arduino\hardware\avr\1.6.11\cores\arduino\CDC.cpp:129:34: error: 'MAGIC_KEY' was not declared in this scope
     *(uint16_t *)magic_key_pos = MAGIC_KEY;
```
I have to go all the way back to Arduino AVR Boards 1.6.10 before it will compile.

facchinm, 2016-09-26:
I believe this can be mitigated by adding `__attribute__((used))` to the definitions of `osc` (and any other variable that gets purged by LTO).

The actual behaviour is correct, since that variables are not "exported" but externally used and only resolved at linkage time (not really smart since it depends on `ld` execution order).
We should probably fix the library if it wasn't in `RETIRED` state.

The second bug is totally unrelated, if you want to solve it manually you should patch the core with https://github.com/arduino/Arduino/commit/a73c18c1f4656f356f979dfdf9c21695d7126890

facchinm, 2016-09-26:
Patched here https://github.com/arduino-libraries/Robot_Control/commit/7eebf7fb07f72597fe810b68239b7d1526fe258f.
We are not releasing the updated library since it is retired but you can grab the zip and place it in the sketchbook so it overrides the bundled one

per1234, 2017-01-15:
@facchinm don't you think it would be a good idea to un-retire the Arduino Robot libraries and ship the fixed version of Robot Control now that arduino.org is using this IDE also. They are selling the Arduino Robot on arduino.org. There were two reports of people not being able to compile the Robot Control library on the forum in the last 24 hours alone. I don't see what the point is of shipping known broken code with the Arduino IDE.

facchinm, 2017-01-16:
Correct, I'm reopening this issue to keep track of the problem, we are solving it via Lib manager fists and then bundle the updated library in the next IDE release. Thanks for pointing that out!

per1234, 2017-07-06:
Robot Control 1.0.3 is bundled with Arduino IDE 1.8.2 and newer so I think this can be closed now. Thanks!

closed 2017-07-06 by per1234