arduino/Arduino #5418: Inconsistent error highlight behavior on first line of sketch/tab

per1234, 2016-09-26:
Using Arduino IDE 1.6.13 Hourly Build 2016/09/23 03:33 with Windows 7 64 bit

- Create a sketch with an error on line 1.
- **Sketch > Compile/Verify** - line 1 is highlighted.
- Move the cursor to line 1.
- Press Enter.
- The error line moves to line 2 but the error highlight remains at line 1.
- **Sketch > Compile/Verify** - line 2 is highlighted.
- Move the cursor to line 1.
- Press Enter.
- The error line moves to line 2 and the error highlight also moves to line 2, the correct and normal behavior.

This happens on any tab.

matthijskooijman, 2016-09-27:
I think this is a "feature" of rsyntaxarea or swing. When a highlight is set to position 0 (== before first char), inserting a new line before the first line is interpreted as inserting a newline *after* position 0, so the highlight does not move. I believe this only applies for the first line, to make sure you can anchor a highlight to the top of a file.

The Java documentation seems to confirm this. From the [javax.swing.text.Position documentation](https://docs.oracle.com/javase/7/docs/api/javax/swing/text/Position.html):

> Position with an offset of 0 is a special case. It never changes its offset while document content is altered.

One way to prevent this would be to use offset 1 for the first line instead of 0, but that would require changes in RSyntaxTextArea [here](https://github.com/bobbylight/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rtextarea/LineHighlightManager.java#L59-L60) or [here](https://github.com/bobbylight/RSyntaxTextArea/blob/master/src/main/java/org/fife/ui/rtextarea/LineHighlightManager.java#L194-L197).

matthijskooijman, 2016-09-27:
I also think this is unrelated to the editor refactor, I think this has been working like this for a lot longer (also seeing that @per1234 did not mention this working on earlier versions).

On an entirely different note, @per1234 thanks for all your testing and excellent bug reports. Much appreciated!

per1234, 2016-09-27:
That's correct, this issue occurs all the way back to IDE 1.6.5-r5. 1.6.4 was the first version to highlight the error line but it actually selects the line which doesn't allow this issue to occur since it just erases the line if you type anything while it's selected. I don't think it's a huge problem worth much effort to fix. I thought it might be something as simple as replacing a 1 with a 0 in the code so I decided to report it anyway.

I'm glad if my bug reports are helpful. Thanks for being so responsive and understanding of my clumsiness. I've actually learned a lot from reading the comments and commits in these repositories. I've especially noticed that you take the time to do a bit of teaching along the way @matthijskooijman.

matthijskooijman, 2017-01-19:
I've forwarded this bug to RSyntaxArea, let's see what they think.