arduino/Arduino #5419: MacOS 10.12 kernel Panic when any Arduino board attached

thomasb95, 2016-09-26:
kernel Panic when any Arduino board attached  and the IDE does not have to be launched. Tried both 
USB 3.0 ports happens about 5 seconds after plugging in the device. Placed a monitor between the USB port and Arduino the board seems to not put any
unusual stress on the port.
First morning Panic:
Anonymous UUID:       E5A41AF7-787C-7848-72D6-E9703849CA10

Sat Sep 24 20:15:22 2016

*** Panic Report ***
panic(cpu 5 caller 0xffffff8003e09f4a): Kernel trap at 0xffffff8003d4182c, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff81e9f498fe, CR3: 0x000000000744c000, CR4: 0x00000000001627e0
RAX: 0xffffff81ee3b3000, RBX: 0x0000000000000000, RCX: 0xfffffffffbb968e8, RDX: 0xffffff80044cd5d0
RSP: 0xffffff81ee31bc30, RBP: 0xffffff81ee31bd00, RSI: 0xffffff81ee3b3afe, RDI: 0xffffff800446a0e0
R8:  0xffffff7f86ced728, R9:  0x0000000000000000, R10: 0xffffff8004520a68, R11: 0xffffff81e199d058
R12: 0xffffff803060ca00, R13: 0xffffff800446a0e0, R14: 0x0000000000000000, R15: 0xffffff81ee3b3afe
RFL: 0x0000000000010297, RIP: 0xffffff8003d4182c, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff81e9f498fe, Error code: 0x0000000000000000, Fault CPU: 0x5, PL: 0

Backtrace (CPU 5), Frame : Return Address
0xffffff81ee31b8c0 : 0xffffff8003cf748c 
0xffffff81ee31b940 : 0xffffff8003e09f4a 
0xffffff81ee31bb20 : 0xffffff8003ca9743 
0xffffff81ee31bb40 : 0xffffff8003d4182c 
0xffffff81ee31bd00 : 0xffffff8004286fe6 
0xffffff81ee31bd20 : 0xffffff7f849fd00a 
0xffffff81ee31bd70 : 0xffffff7f8498a00a 
0xffffff81ee31bd90 : 0xffffff7f8498ff9d 
0xffffff81ee31be10 : 0xffffff80042c00ba 
0xffffff81ee31be80 : 0xffffff7f8498f8dd 
0xffffff81ee31bed0 : 0xffffff7f849947e1 
0xffffff81ee31bf10 : 0xffffff7f8499453f 
0xffffff81ee31bf30 : 0xffffff80042bd621 
0xffffff81ee31bf80 : 0xffffff80042bcc06 
0xffffff81ee31bfb0 : 0xffffff8003ca6af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.1)[6A671CD8-5527-3A10-8675-1421D158D7A7]@0xffffff7f84959000->0xffffff7f849c0fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[DB526B45-1A45-3A81-A0C1-57F826CADEDF]@0xffffff7f8494c000
         com.apple.iokit.IOUSBFamily(900.4.1)[8F6207EC-608D-373A-B35E-E6578202F58D]@0xffffff7f849c9000->0xffffff7f84a61fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[731443D8-78D5-30C8-939A-1ED3E857CA22]@0xffffff7f84532000
            dependency: com.apple.iokit.IOUSBHostFamily(1.1)[6A671CD8-5527-3A10-8675-1421D158D7A7]@0xffffff7f84959000

BSD process name corresponding to current thread: kernel_task
Boot args: b
System Profile:
Graphics: Intel Iris Pro, Intel Iris Pro, Built-In
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x134), Broadcom BCM43xx 1.0 (7.21.171.10.1a16)
Bluetooth: Version 5.0.0f18, 3 services, 27 devices, 1 incoming serial ports
PCI Card: pci11c1,5901, IEEE 1394 Open HCI, Thunderbolt@195,0,0
PCI Card: Apple 57762-A0, Ethernet Controller, Thunderbolt@10,0,0
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.1
Thunderbolt Device: Thunderbolt to Gigabit Ethernet Adapter, Apple Inc., 1, 5.5
Thunderbolt Device: Thunderbolt to FireWire Adapter, Apple Inc., 3, 5.6
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1600 MHz, 0x80AD, 0x484D54343147533641465238412D50422020
USB Device: USB 3.0 Bus
USB Device: USB3.0 Hub
USB Device: USB3.0 Card Reader
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 2.0 HUB

USB Device: Expansion Desk
USB Device: USB2.0 Hub
USB Device: iPhone
USB Device: iPad
USB Device: USB 2.0 Hub [MTT]
USB Device: SimpleDrive USB 2.0
USB Device: ORTEK KH12 USB Hub
USB Device: USB Hub/Keyboard
USB Device: Ext HDD 1021
USB Device: My Book
USB Device: USB 2.0 Hub
USB Device: USB2.0 Hub
USB Device: Cruzer
USB Device: Cruzer
Serial ATA Device: APPLE SSD SM0512F, 500.28 GB
Network Service: Thunderbolt Ethernet, Ethernet, en3
Network Service: Parallels Shared Networking Adapter #0, Ethernet, vnic0
Model: MacBookPro11,2, BootROM MBP112.0138.B17, 4 processors, Intel Core i7, 2 GHz, 16 GB, SMC 2.18f15

Report for all subsequent Panics and panic took about 2 seconds:
Anonymous UUID:       E5A41AF7-787C-7848-72D6-E9703849CA10

Sun Sep 25 20:43:09 2016

*** Panic Report ***
panic(cpu 4 caller 0xffffff801f609f4a): Kernel trap at 0xffffff801f54182c, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0xffffff91ffffff56, CR3: 0x0000000022c4c000, CR4: 0x00000000001627e0
RAX: 0xffffff920af9b000, RBX: 0x0000000000000000, RCX: 0xfffffffff5064f40, RDX: 0xffffff801fccd5d0
RSP: 0xffffff923316bc30, RBP: 0xffffff923316bd00, RSI: 0xffffff920af9bafe, RDI: 0xffffff801fc6a0e0
R8:  0xffffff7fa24ed728, R9:  0x0000000000000000, R10: 0xffffff801fd20a68, R11: 0xffffff81fd0be058
R12: 0xffffff8040a5e000, R13: 0xffffff801fc6a0e0, R14: 0x0000000000000000, R15: 0xffffff920af9bafe
RFL: 0x0000000000010283, RIP: 0xffffff801f54182c, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff91ffffff56, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff923316b8c0 : 0xffffff801f4f748c 
0xffffff923316b940 : 0xffffff801f609f4a 
0xffffff923316bb20 : 0xffffff801f4a9743 
0xffffff923316bb40 : 0xffffff801f54182c 
0xffffff923316bd00 : 0xffffff801fa86fe6 
0xffffff923316bd20 : 0xffffff7fa01fd00a 
0xffffff923316bd70 : 0xffffff7fa018a00a 
0xffffff923316bd90 : 0xffffff7fa018ff9d 
0xffffff923316be10 : 0xffffff801fac00ba 
0xffffff923316be80 : 0xffffff7fa018f8dd 
0xffffff923316bed0 : 0xffffff7fa01947e1 
0xffffff923316bf10 : 0xffffff7fa019453f 
0xffffff923316bf30 : 0xffffff801fabd621 
0xffffff923316bf80 : 0xffffff801fabcc06 
0xffffff923316bfb0 : 0xffffff801f4a6af7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOUSBHostFamily(1.1)[6A671CD8-5527-3A10-8675-1421D158D7A7]@0xffffff7fa0159000->0xffffff7fa01c0fff
            dependency: com.apple.driver.AppleBusPowerController(1.0)[DB526B45-1A45-3A81-A0C1-57F826CADEDF]@0xffffff7fa014c000
         com.apple.iokit.IOUSBFamily(900.4.1)[8F6207EC-608D-373A-B35E-E6578202F58D]@0xffffff7fa01c9000->0xffffff7fa0261fff
            dependency: com.apple.iokit.IOPCIFamily(2.9)[731443D8-78D5-30C8-939A-1ED3E857CA22]@0xffffff7f9fd32000
            dependency: com.apple.iokit.IOUSBHostFamily(1.1)[6A671CD8-5527-3A10-8675-1421D158D7A7]@0xffffff7fa0159000

BSD process name corresponding to current thread: kernel_task
Boot args: b

facchinm, 2016-09-26:
HI @thomasb95 ,
did it start all of a sudden after updating to 10.12? Which board(s) can you test with? Also, if you could post your Mac model it would be great

thomasb95, 2016-09-26:
Yes Right after the update. 
Change for time to panic 5 sec. I one disconnects fast enough
the system just goes through a forced reset otherwise it goes into a kernel Panic.

Mega 2560, Nano 5v and 3.3v, Uno R3, Leanardo, Trinket 5v and 3.3, Gemma, Raspberry 2 and 3 and a Pro mini 328P (need to still figure a way to upload since I am on a fixed income.) 

  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro11,2
  Processor Name:	Intel Core i7
  Processor Speed:	2 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	6 MB
  Memory:	16 GB
  Boot ROM Version:	MBP112.0138.B17
  SMC Version (system):	2.18f15
  Hardware UUID:	308E35AF-BFFE-5020-B9D9-931C49C8577B

USB 3.0 Bus:

  Host Controller Driver:	AppleUSBXHCILPTH
  PCI Device ID:	0x8c31 
  PCI Revision ID:	0x0005 
  PCI Vendor ID:	0x8086 

USB3.0 Hub             :

  Product ID:	0x0812
  Vendor ID:	0x2109  (VIA Labs, Inc.)
  Version:	b.e1
  Speed:	Up to 5 Gb/sec
  Manufacturer:	VIA Labs, Inc.         
  Location ID:	0x14500000 / 4
  Current Available (mA):	900
  Current Required (mA):	0
  Extra Operating Current (mA):	0

Apple Internal Keyboard / Trackpad:

  Product ID:	0x0262
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	2.25
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14400000 / 3
  Current Available (mA):	500
  Current Required (mA):	40
  Extra Operating Current (mA):	0
  Built-In:	Yes

BRCM20702 Hub:

  Product ID:	0x4500
  Vendor ID:	0x0a5c  (Broadcom Corp.)
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14300000 / 2
  Current Available (mA):	500
  Current Required (mA):	94
  Extra Operating Current (mA):	0
  Built-In:	Yes

Bluetooth USB Host Controller:

  Product ID:	0x8289
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.26
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x14330000 / 8
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0
  Built-In:	Yes

USB 2.0 HUB
      :

  Product ID:	0x2812
  Vendor ID:	0x2109  (VIA Labs, Inc.)
  Version:	85.70
  Speed:	Up to 480 Mb/sec
  Location ID:	0x14200000 / 5
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

USB2.0 Hub:

  Product ID:	0x0610
  Vendor ID:	0x05e3  (Genesys Logic, Inc.)
  Version:	92.22
  Speed:	Up to 480 Mb/sec
  Manufacturer:	GenesysLogic
  Location ID:	0x14230000 / 12
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

USB 2.0 Hub [MTT]:

  Product ID:	0x0201
  Vendor ID:	0x1a40  (TERMINUS TECHNOLOGY INC.)
  Version:	1.00
  Speed:	Up to 480 Mb/sec
  Location ID:	0x14220000 / 11
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

ORTEK KH12 USB Hub:

  Product ID:	0x9735
  Vendor ID:	0x05a4  (Ortek Technology, Inc.)
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	ORTEK
  Location ID:	0x14226000 / 16
  Current Available (mA):	500
  Current Required (mA):	100
  Extra Operating Current (mA):	0

USB Hub/Keyboard:

  Product ID:	0x9735
  Vendor ID:	0x05a4  (Ortek Technology, Inc.)
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	ORTEK
  Location ID:	0x14226100 / 21
  Current Available (mA):	500
  Current Required (mA):	0
  Extra Operating Current (mA):	0

One more thing. My response maybe slow sometimes since I'm blind. 

My Background Navy ET, Systems ENG and QA Fermware Programmer. 

Past Fields of work Radar surface and air, Crypto, Nave aids Tech, Old style mini computers (Store Level), Motion Control, Certified Apple Technician and SE,Good Background in UNIX FreeBSD, NetBSD, and Darwin, Linux.  QA Micom 4400, 4450 4460 and Nortel 6400 and QA Spirent products. Languages C, C++, Fortrain. BA in Physics.

Hope this Helps

PaoloP74, 2016-09-29:
For Arduino Clones, try to update CH340/341 driver --> http://www.wch.cn/download/CH341SER_MAC_ZIP.html

martinlall, 2016-10-07:
macOS Sierra, Mega clone, same panic. Updating drivers didn't help. 
Macbook Pro 2014

tofifee, 2016-10-08:
Ha! Same here running a hack. Started after an update in the Arduino IDE.

closed 2017-01-19 by facchinm