arduino/Arduino #5420: Hacking/BuildProcess documentation needs update

matthijskooijman, 2016-09-26:
The page https://www.arduino.cc/en/Hacking/BuildProcess contains some things that are outdated or need updated. From a first quick read, I noticed:
 - "The include path includes the sketch's directory,", I do not think this is, or was ever true. Or perhaps it is true only for sketches, not libraries and cores?
- "These are inserted after any comments or pre-processor statements", I think these are inserted just before the first function now.
 - "Finally, the contents of the current target's main.cxx file are appended to the bottom of your sketch.", this is no longer true for a long time.
 - Recursive compilation in the `src/` subdirectory is not mentioned.
 - Files "without extension" are mentioned, but it would be good to state that these are without extension in the IDE, but use .ino in the filesystem.

There might be other issues, I haven't thoroughly reviewed them yet.

It is probably good to link to this page from https://www.arduino.cc/en/Guide/Environment#toc8 as well.

per1234, 2016-09-26:
It seems problematic to have a Build Process page on the website and one on the GitHub wiki(https://github.com/arduino/Arduino/wiki/Build-Process) since it doubles the maintenance effort to keep both up to date. I prefer a wiki page since it's editable by anyone and the GitHub wiki doesn't seem to incur much spam/vandalism but either way I think one of the two should be deleted after preserving any useful content.

akash73, 2017-06-05:
➤ Andrea Richetta commented:

HI Cristian,
I suggest to keep only one page updated, seems that the one on github is more reliable. should we add a link on the original page on arduino.cc and redirect users to the github page?

akash73, 2017-06-05:
➤ Cristian Maglie commented:

I think it's ok to remove the page on arduino.cc, I'd just keep the notice at the beginning:

per1234, 2017-07-09:
I've done some work on https://github.com/arduino/Arduino/wiki/Build-Process that fixes all of the issues that @matthijskooijman mentioned except for:
>Recursive compilation in the src/ subdirectory is not mentioned.

because I don't really know how to explain that.

I took a look at https://www.arduino.cc/en/Hacking/BuildProcess and it seems inferior to https://github.com/arduino/Arduino/wiki/Build-Process in every way. I see nothing of value there that needs to be transferred over. So I'd say go ahead with removing the all content from https://www.arduino.cc/en/Hacking/BuildProcess and add a link to https://github.com/arduino/Arduino/wiki/Build-Process.

akash73, 2017-07-17:
ok content moved to https://github.com/arduino/Arduino/wiki/Build-Process
thanks

closed 2017-07-17 by akash73