arduino/Arduino #5423: Can not save sketches on non admin Windows user account

j54n1n, 2016-09-28:
Hello,

With the latest version of the Arduino IDE v1.6.12 I get the following message when I try to save (Ctrl+Shift+S) a sketch with its default name. The error message says something like that the sketch can not be saved in the current sketchbook since it is already exists already, but that is not the case. Afterwards if I try to save the sketch again I get redirected in the Save As window to the TEMP folder of the current user.
In the particular in my case I installed the Arduino IDE from the ZIP file (non admin install) within the current non admin user folder.
The same issue does not happen if I use the Arduino IDE v1.6.11 ZIP file.

sandeepmistry, 2016-09-28:
This also impacts OS X. If I create a new sketch and save, the following is shown:

[image]

The sketch is not saved in my sketch book folder, but rather in `/var/folders/m9/y22yzb0d49x_kws_6jtwdphw0000gn/T/untitled617461295.tmp/sketch_sep28b/sketch_sep28b.ino`.

cmaglie, 2016-09-29:
Thank you for the bug report!

the fix will be available on IDE 1.6.13 or next hourly build

closed 2016-09-29 by cmaglie

matthijskooijman, 2016-09-29:
@cmaglie, thanks for fixing my oopsies :-)

Fix looks good to me.

commit 5c097c03 (2016-09-29): Fixed regression: trivial error on String comparison

Fix #5423

The regression has been introduced in:
72f815b Refactor file adding and renaming, and save as handling