arduino/Arduino #5424: Arduino  1.6.12 / Travis CI: “Selected library is not available”

krzychb, 2016-09-28:
Hi,

I have updated previously working Travis CI script by changing Arduino IDE version from 1.6.9 to 1.6.12.

As result in the line `- arduino --install-library "WebSockets"`, the script is failing with `Selected library is not available` - https://travis-ci.org/krzychb/EspScopeA0/builds/163400605

Troubleshooting:

  1.	The same error is reported if I select Arduino 1.6.11 - https://travis-ci.org/krzychb/EspScopeA0/builds/163373114
  2.	If I change Arduino back to 1.6.9 then the error is `java.lang.Exception: CRC doesn't match. File is corrupted` (checked doing the build three times).
  3.	The same script was passing without any issues about 16 days ago - https://travis-ci.org/krzychb/EspScopeA0/builds/159405818
  4.	There are no any problems loading a fresh copy of “WebSockets” library on my PC using Library Manager in Arduino IDE 1.6.9, 1.6.11 and 1.6.12.
  5.	I have checked using Travis CI script with some other libraries and see the same issue.

Could you give me some guidance what the issue may be / how to troubleshoot it?

Krzysztof

cmaglie, 2016-10-03:
@krzychb 
it seems that the library_index.json is downloaded but not used right awa (so the previous, empy/missing, is used).

As a quick workaround I found this:
https://github.com/cmaglie/test-ci/commit/6b4d02a57a2dd4f2093b04affd5fe8eb2a4ce866

I'm investigating why the updated `library_index.json` is not used.

krzychb, 2016-10-03:
@cmaglie, 
I have added the workaround to my repository and it works!
Thanks a lot :+1:

commit a8525c6b (2016-10-04): Parse libraries index after updating it (when launche from CLI)

Fix #5424
closed 2016-10-04 by cmaglie