arduino/Arduino #5430: Board upgrade released 26 Sept 2016 broke avrdude

Padoup-Padoup, 2016-09-30:
Hi all,

I am experiencing a blocking bug since the last update of the Arduino IDE. It is the same as described on [this forum](http://forum.arduino.cc/index.php?topic=426318.0). I copy-pasted below the relevant error message:

```{shell}
Une erreur est survenue lors du transfert du croquis
/home/maxime/.arduino-1.6.12/hardware/tools/avr/bin/avrdude: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.14' not found (required by /home/maxime/.arduino-1.6.12/hardware/tools/avr/bin/avrdude)
/home/maxime/.arduino-1.6.12/hardware/tools/avr/bin/avrdude: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.15' not found (required by /home/maxime/.arduino-1.6.12/hardware/tools/avr/bin/avrdude)
```

Note that:

- It is definitely related with the update (I was using the IDE, compiling and transferring without issues when the update happened).
- The compilation step works fine, the error message occurs when trying to upload the sketch.

Versions:

- Arduino: 1.6.12 (installed from the tar.xz package)
- Debian Wheezy  64 bits.

Is there something obvious I might be missing? 

Thank you!

facchinm, 2016-09-30:
Ooops, thanks for spotting this. Our jenkins job builds avrdude on Ubuntu12.04 VMs, which ships a newer libc compared to debian7. I'm rebuilding them on Debian7 slaves and I believe we'll ship the fixed tools in a future release of AVR core.
Thanks for reporting! In the meantime, as soon as the job completes, I'll post the links for the updated package so you can manually replace it.

facchinm, 2016-09-30:
[avrdude-6.3.0-arduino6-x86_64-pc-linux-gnu-debian7.zip](https://github.com/arduino/Arduino/files/502533/avrdude-6.3.0-arduino6-x86_64-pc-linux-gnu-debian7.zip)

Padoup-Padoup, 2016-09-30:
Great, this solves the issue in my case! Feel free to close the issue when the fix is released! Many thanks!

matthijskooijman, 2016-12-06:
@facchinm, did this get released already?

facchinm, 2016-12-06:
Not yet, it will be included in next avr core release

closed 2017-02-28 by facchinm