arduino/Arduino #5431: Path+Sketch length error / Could not create sketch / Failed to open sketch / Only showing include files

degerrit, 2016-09-30:
Not sure why this issue appeared (maybe the Arduino sketch folder was moved in the last months?), but Arduino IDE on Windows is now having issues with Sketches that have long path+filenames. 

Other programs, i.e. Wordpad, can open the .ino files without any problems, so it's not an OS limitation.

Examples:
D:\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1.ino
105 chars - result: **opens OK**

D:\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1_Dist12_13_8s_s\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1_Dist12_13_8s_s.ino
135 chars - result: **Not OK, "Could not create sketch"**

Interestingly, the first sketch I opened after upgrading to 1.6.12 had some include files with short filenames (x10rf.h, x10rf.cpp) - the result here is NO ERROR MESSAGE, but the IDE opening only the include files, and not the .ino file.

I tried downgrading back to 1.6.9, but this didn't help for me.
---
Default path - looks OK:
-----
[image]

Wordpad - no issue:
-----
[image]

The long path
-----
[image]

Long (>128 character?) .INO sketch file that contains include files - IDE only shows the short include files, NO ERROR message is given
-----
[image]

Short test sketch with no includes, but long path - ERROR
-----
[image]

-----
[image]

-----
[image]

Same sketch, renamed / shortened path and Sketch name - opens OK
-----
[image]

degerrit, 2016-10-01:
Updated my original bug, a downgrade to 1.6.9 via full uninstall/reinstall **resolves** the problem. 
Previously my downgrade to 1.6.9 that didn't resolve the problem involved swapping the Arduino folder for a backed-up folder, not a full uninstall/reinstall. Very strange!

degerrit, 2016-10-01:
OK, 1.6.11 also does not have this problem. Maybe it's the editor code refactoring in 1.6.12

closed 2016-10-03 by cmaglie

cmaglie, 2016-10-03:
Thanks for the bug report.

In this case is not the full path length the problem, but only the **sketch name** length.

The sketch name is limited to 63 chars and this limit has always been there but, before 1.6.12, the check was made only on the sketch name **without the .ino extension**. After 1.6.12 this check has been changed to test the full file name (with .ino extension). So a sketch that previously fitted exactly the 63 chars limit after 1.6.12 it will suddenly became invalid because the .ino extension will made it of 64+4 -> 67.

Now I've fixed it by making the check again on the sketch name. Of course this bug affects only sketch with very long names (nearing 63 chars).

Lizews2006, 2019-10-27:
> Not sure why this issue appeared (maybe I moved the Arduino sketch folder - I hadn't used my Arduino in a few months), but my Arduino IDE on Windows is now having issues with Sketches that have long path+filenames. I created these in Arduino IDE, so previously I had no issues with this.
> 
> Other programs, i.e. Wordpad, can open the .ino files without any problems, so it's not an OS limitation.
> 
> Examples:
> D:\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1.ino
> 105 chars - result: **opens OK**
> 
> D:\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1_Dist12_13_8s_s\Blink_undDet-D8_Water-A0_Temp-D2_X10RF-D4_LDR-A1_Dist12_13_8s_s.ino
> 135 chars - result: **Not OK, "Could not create sketch"**
> 
> Interestingly, the first sketch I opened after upgrading to 1.6.12 had some include files with short filenames (x10rf.h, x10rf.cpp) - the result here is NO ERROR MESSAGE, but the IDE opening only the include files, and not the .ino file.
> 
> ## A full downgrade back to 1.6.9 (uninstall/reinstall) fixes the issue.
> ## Default path - looks OK:
> [image]
> 
> ## Wordpad - no issue:
> [image]
> 
> ## The long path
> [image]
> 
> ## Long (>128 character?) .INO sketch file that contains include files - IDE only shows the short include files, NO ERROR message is given
> [image]
> 
> ## Short test sketch with no includes, but long path - ERROR
> [image]
> 
> [image]
> 
> [image]
> 
> ## Same sketch, renamed / shortened path and Sketch name - opens OK
> [image]

Same here. If anyone found a way, pls email me: lizews2006@gmail.com
You see, I'm actually a student and I need to finish the code but I cannot open it. The competition is in one week time, so pls help me... Thanks!

cmaglie, 2019-10-30:
@Lizews2006 thanks for the bug report but this issue should be solved with **IDE 1.8.10**, if not please open another issue and do not comment on already closed issues.

Anyway, if you're in a hurry, you may consider renaming the `.ino` file **and the containing folder** to a shorter name for a quick workaround.

commit 653a0527 (2016-10-03): Check sanitaryName only on basename without extension

This regression originates from:

8725bb1e Clean up sketch loading

before this commit the sketch name sanitization was made on the sketch
name without the extension.
After 8725bb1e instead the name sanitization is made on the filename, so
including the ".ino" extension.

This lead to a weird corner case, caused by the limit of 63 characters
on the sketch name: before 8725bb1e it would be possible to save a sketch
with a name of exactly 63 characters, but after 8725bb1e this sketch will
suddenly becomes invalid becuase the 63 chars name +  [... cut]