arduino/Arduino #5435: Upgrade from v1.6.10 to v1.6.12 - esp8266 json and boards.txt error

electronicsguy, 2016-10-01:
Hello,
      I am facing boards error again, after upgrading Arduino from v1.6.10 to v1.6.12. I had faced a similar error before. 

The installation of the IDE itself went perfectly. When I opened it up, it said something like the AVR library is out of date, blah blah. So I tried to upgrade. It complains about a missing boards.txt and not able to download the esp8266 json file.

Why do we keep getting these errors everything we upgrade? Shouldn't this be detected in testing? There was no customizations in my setup either.
[image]

shiftleftplusone, 2016-10-01:
as there is supposed to be no reasonable beta testing and controlling before releasing an update, the best wayfor end users IMO is simply not to upgrade.
Perhaps at least to the latest version before the last update.

Sheep666, 2016-10-02:
Did you deal with ?

shiftleftplusone, 2016-10-02:
no, I still use 1.6.5, in the moment no reason why to change that.
Since in a later release Java VM had been changed, everything was messed up with my programs, no proprietary libs and no Due board worked any longer. So I reinstalled Windows from an image (anything else failed)  and reinstalled 1.6.5 again - everything worked smooth again then .
That experience healed me for the entire future.

about a missing  " reasonable beta testing and controlling before releasing" I meanwhile have another throrie:
Releasing a new version actually IS the beta test, started and performed by end users who try to dealwith it.
;)

facchinm, 2016-10-03:
Hi @electronicsguy ,
the issue you are encountering is due to an old AVR core not being installed correctly. With IDE 1.6.12 we bundled AVR core 1.6.14 . The fact that the Board Manager shows 1.6.13 with the "no board txt" error message means that the 1.6.13 folder is empty. 
Since IDE 1.6.11 bundled this version, the IDE was not complaining as it was using the bundled one. Delete that (empty) folder and you'll be ok.

I agree that we should check the installed/uninstalled cores in a more reliable way, I'm not closing this issue so we can keep track of all the corner cases a user could encounter.

The esp8266 json error is, instead, caused by connectivity problems, probably on their side. The json look reachable for me at the moment, reopening the board manager should do the trick.

electronicsguy, 2016-10-03:
@facchinm   Thanks for the update... Appreciate the work you guys do. I feel that coming up with some kind of a standard for writing libraries/board files should be adopted (if it isn't already). If it already exists, it doesn't seem to be up to the mark. Otherwise, these kind of issues will keep coming up in the future, as you make changes to Arduino.

facchinm, 2016-10-03:
For both the libraries and the core we have an extensive documentation here on Github [wiki](https://github.com/arduino/Arduino/wiki/_pages) (unluckily Google rank of these pages is really low so they are almost invisible during a normal search). 

The installing/uninstalling problems arise from our implementation faults, and the fact that the folder containing the cores is inside an hidden folder doesn't help. 

What we really miss is a way to keep track of the state of the installed boards, because right now the only way to discover if a core is installed or not is checking for the existence of a folder (which could be empty due to a faulty previous uninstallation). We are working on it, so future releases will be more and more reliable :smile: