arduino/Arduino #5436: New signatures to void ESP8266WebServer::sendContent

wellrats, 2016-10-01:
Hi all,
Would it be possible in next version of this lib insert another signature for
ESP8266WebServer::sendContent as follow

```
void ESP8266WebServer::sendContent(const char* content, size_t len);
void ESP8266WebServer::sendContent(const char* content)
```

It's very usefull when you are parsing a HTML file, like a **template**, and sending little pieces of this file to client. As length is set to ```CONTENT_LENGHT_UNKNOW```, chunked encoding is used, so call ``` <server_instance>.client().write( ... )``` directly is not an option.

Thanks in advance

```
void ESP8266WebServer::sendContent(const char* content, size_t len) {
  const char * footer = "\r\n";
  if(_chunked) {
    char * chunkSize = (char *)malloc(11);
    if(chunkSize){
      sprintf(chunkSize, "%x%s", len, footer);
      _currentClient.write(chunkSize, strlen(chunkSize));
      free(chunkSize);
    }
  }
  _currentClient.write(content, len);
  if(_chunked){
    _currentClient.write(footer, 2);
  }
}

void ESP8266WebServer::sendContent(const char* content) {
  sendContent(content, strlen(content));
}

void ESP8266WebServer::sendContent(const String& content) {
  sendContent(content.c_str(), content.length());
}
```

luc-github, 2016-10-01:
Actually you do not need to specify size when sending if you use `setContentLength(CONTENT_LENGTH_UNKNOWN);` you just need to finish your sending by `sendContent("");`

you can see how it can be used with template on my project : https://github.com/luc-github/ESP3D/blob/master/esp3d/webinterface.cpp#L214-L394

wellrats, 2016-10-01:
Wrong place ... sorry

closed 2016-10-01 by wellrats