arduino/Arduino #5437: https downloads

NicoHood, 2016-10-03:
A lot of link/downloads in the source do use http for download while a secure https is available instead.
Those files can be found for example via github:
https://github.com/arduino/Arduino/search?utf8=%E2%9C%93&q=http%3A%2F%2F

Most important are the sources in build.xml:
https://github.com/arduino/Arduino/blob/master/build/build.xml

And also the avr cores:
https://github.com/arduino/Arduino/blob/master/hardware/package_index_bundled.json

Since arduino has https encryption you could possibly replace verything starting with `http://www.arduino.cc` with https. That would cover most links and would not break anything.

cmaglie, 2016-10-03:
All files downloaded from build.xml or from json indexes are verified with checksums (sha/sha256).
Changing everything to https will increase the servers load without any improvement in security.

Maybe there are some places where using to https will improve security but this is not in build.xml or inside the indexes.

NicoHood, 2016-10-03:
Its not about integrity, its about confidentiality (but still happy to see that some people understand the meaning of integirty finally =) ).

I personally do not want everyone to see what I download, no matter if its "just" arduino or not. I am not sure about sever loads, but https is the (... not future, actually the past) ... its time to use https. The server should be able to handle such downloads.

NicoHood, 2016-11-26:
Please conside the switch to https. Archlinux now switched every possible source to https and gpg signatures. And if i am right debian also has a policy about that:
https://www.archlinux.org/todo/use-gpg-signatures-and-https-sources/

Here is what google thinks about https:
https://developers.google.com/web/fundamentals/security/encrypt-in-transit/why-https

Everything is already covered in this PR:
https://github.com/arduino/Arduino/pull/5438

And the gpg signing is tracked here:
https://github.com/arduino/Arduino/issues/5619