arduino/Arduino #5439: MCP9802 Driver: New Library Contribution

nadavmatalon, 2016-10-03:
Hi,

I created (and tested with hardware) a comprehensive library for driving the MCP9802 (12-BIT Temperature Sensor with Hysteresis, Alert, and HW I2C interface).

The library comes complete with testing & usage sketch examples and functional documentation.

The current version of the library is: v1.0.0 (for which a 'Github Release' has been created)

The Github repo of the library can be found at: https://github.com/nadavmatalon/MCP9802

Would be great if it could be added to the list of contributed libraries.

Any feedback/suggestions/bug reports would, of course, be welcome.

Thanks,
Nadav

agdl, 2016-10-04:
Done thank you!

closed 2016-10-04 by agdl

nadavmatalon, 2016-10-05:
Cool, thanks agdl!

A couple of quick follow-up question if I may: 

- Does the list get updated every fixed period? (I'm asking as I can't see the library on the Playground)

- between then and now, I released a couple of minor versions so the latest version is 1.2.0. Is there anything that needs to be updated somewhere in this context (apart from the Github repo which is up to date of course)?

- Is it now possible to install the library via the 'Library Manager' in the Arduino IDE (>1.5)?

All the best,
Nadav

agdl, 2016-10-07:
Everytime you post an upated release it will be available in the library manager.
Obviously your library.properties files must be correct.

The libarry is not posted on playground but only in the library manager