arduino/Arduino #5441: .hh and .hpp files in libraries can only be included after a .h file from that library

per1234, 2016-10-04:
Using Arduino IDE 1.6.13 Hourly Build 2016/10/03 06:51 with Windows 7 64 bit

https://github.com/arduino/Arduino/pull/3186 added support for including .hpp and .hh files from libraries.

If I have a library installed with the folder structure:

foo
|_foo.h
|_foo.hh
|_foo.hpp

With Arduino IDE 1.6.6 or newer, doing:
```arduino
#include <foo.hh>
```
or
```arduino
#include <foo.hpp>
```
causes compilation to fail:
```
fatal error: bar.hh: No such file or directory
 #include <foo.hh>
```
It will only compile if I do:
```arduino
#include <foo.h>
#include <foo.hh>
```
Using Arduino IDE 1.6.5-r5 I don't need to include the .h file.

facchinm, 2016-10-04:
@per1234 , could you move it to `arduino-builder` repo? Thx

per1234, 2016-10-04:
OK, I've merged with https://github.com/arduino/Arduino/issues/5442 and moved to the arduino-builder repo: https://github.com/arduino/arduino-builder/issues/181. Sorry about that, I should have realized the include stuff was arduino-builder.

closed 2016-10-04 by per1234