arduino/Arduino #5447: Recently used  file  feature does not , but  loads multiple files.  IDE 1.6.11.

VaclavSal, 2016-10-04:
This very useful feature loads multiple recently used files - generally one full size window and the next one in standard window size on top of this base window. 
Loading / opening another file requires deleting at least one of these windows because the new file does not load as "top window", but is hidden bellow the full size window. 
The feature should load / display only one true recently used file ( singular) or cascade multiple files loaded so user can select the one of interest - move  as top full size window. 

Opening new file could be done similar way - put the new one in cascade as top window would work nice.  

.

VaclavSal, 2016-10-18:
I like to close this apparently "non issue ".
May I?

facchinm, 2016-10-20:
Sure :smile:
No one else complained about the existing behaviour, so I believe it is ok as it is.
Reopen if needed

closed 2016-10-20 by facchinm