arduino/Arduino #5449: Global system platform.txt

NicoHood, 2016-10-04:
It would be nice to have a platform.txt that cannot be overwritten my the avr-core.
For example I need to change `compiler.path={runtime.tools.avr-gcc.path}/bin/` to `compiler.path=/usr/bin/`. But I also need to add some extra cpp flags that are specific for the avr-gcc 6.x.

I am still not sure whats the best way to solve this, but if we can make the distribution avr-gcc packages more accessible its simpler to package arduino for linux and also especially for archlinux never avr-gcc can already be tested for bugs.

onsky1982, 2017-02-09:
I have the same problem with you. I follow"platform.txt"file link"https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification".I tested "{runtime.ide.path}",found the Global Predefined properties not work.

anno73, 2018-12-22:
Similar here. I'd like to define global hook scripts  (e.g. `recipe.hooks.sketch.prebuild`) without interfering with updates. Maybe located in "sketchbook location\global" or something?